BotoX/xiaomi-m365-firmware-patcher #1: Power trigger sensitivity

Kulimeya, 2018-05-14:
Hi,

thanks for the great solution for such a specific problem, I mentioned your service on /r/xiaomi. 

To my problem: 
Might it be possible to adjust the gas sensitivity in the firmware? The higher power is really great, but the sensitivity is very annoying. With stock firmware the first 80% of the trigger way is very low sensitivity and only the last 20% are pushing really forward, so it is easy to hold low speeds. If you increase the power with the CFW, that sensitivity changes, even if you pulling the trigger only 20% it goes straight forward, which makes it nearly impossible to drive smooth and slow. A solution for that would be really awesome!

Thanks

BotoX, 2018-05-14:
I've also noticed this behavior, the Doc-Hello/Ferry guys from Russia have modified the gas algorithm in their M365Max firmware (CFW and CFW W). But their firmware has an issue that the speed fluctuates by like 3km/h when driving at full speed which is much more annoying than having a little better throttle.

When having to drive slow I make use of the disabled KERS, roll freely and accelerate a little when possible - if this helps you.

Looking into these sorts of complicated things will hopefully be easier when I get my ST-Link from China and can debug the code live as it's running on the scooter.
Right now making sense of this code is not easy as I have no clue what kind of values the variables have at which speed / throttle position etc.

allstone, 2018-05-17:
I have noticed same speed fluctuation going a bit uphill with normal mode max speed while generating FW from 1.3.8 with values DRV138-patched_15_31_20_3_40165_0_1_1 (values in turn 15 kers, 31 normal, 20 eco, 3 start, etc)

oldtrapper, 2018-07-09:
I second the feature request relatet to more sensitivity in throttle switch while on "high power". I am post surgery, handicapped in walking for quite a while but escorting my family with the scooter while they are strolling along, making sightseeintours during our holidays is really fun. Would be even more fun if I could fintune speed or cruisecontrol at speeds between 3 and 6 km/h

RedWavve, 2018-07-12:
> Would be even more fun if I could fintune speed or cruisecontrol at speeds between 3 and 6 km/h

Have you tried to **set your eco mode to 6km/h**, and then checking how the two (power vs speed) throttle algorithm work?

If I remember correctly, the "normal" (speed) algorithm applies a % value of the throttle to the max speed.
- Reasoning here : If eco mode max speed is 6km/h, then 50% throttle might give you 3km/h, etc
- You could then remove eco mode to get 28km/h when you want

**However**, I'm not really sure that the throttle lever is looking for the "eco max speed", and not **ONLY** for the "normal max speed".
- In that case, you would unfortunately have to flash your M365 every time you go on a walk with you family :-(

Please test and tell us about it ;-)

oldtrapper, 2018-07-12:
yess, tried. Not amused. Eco speed set to 6 km/h. Accelerates to about 4 km/h within the first 5 % of the throttlespan. You have to reduce motorstart speed to 2 km/h in order to let engine start at all. Of course you can set max eco speed to 10 or alike but since only the first few mm of throttlespan is used, this is not satisfying. Never tried russian power algorythm but maybe will do so during the next weeks. Reports when there is network coverage :-).

My way for now: Kers removed below 25 km/h and with family cruising in eco mode (max speed set to 25 km/h) ... smooth acceleration and rolling along when throttle released.

jbfuzier, 2018-08-09:
Is there a way to keep default throttle behavior ? Some settings not to change in your app ? Or as soon as you use the custom firmware you looses default throttle behavior?