BotoX/xiaomi-m365-firmware-patcher #3: No option to use customized firmware in the patcher app

waldemar-szostak, 2018-05-23:
First off, many thanks for your efforts already so far!

Secondly, I [created several firmware bin files](https://m365.botox.bz/) but there is no way to flash them into m365: the original (?) app detects the firmware as invalid, and the [patched one](https://cloud.botox.bz/s/PZkfxWjNpgXibT5/download?path=%2F&files=com.m365downgrade-v7_PATCHED.apk) (after repackaging with apktool) - fails to upload the firmware fully - it fails with timeouts.

Could you please advise?

BotoX, 2018-05-23:
The only change in the patched apk is the removal of the hash check.
So if you're having timeouts in the patched APK, then the same thing should also happen in the original one, except if something was fixed in an update.

You could try flashing the original firmware: https://cloud.botox.bz/s/PZkfxWjNpgXibT5?path=%2FUNTOUCHED and see if there is a difference between the apps.

And why repack the patched APK? Works fine for me? com.m365downgrade-v7_PATCHED.apk

waldemar-szostak, 2018-05-23:
I repackaged it, so that I could replace one of the bin files in assets directory (IIRC..) with the firmware I customized via your website.

waldemar-szostak, 2018-05-23:
Thank you! Yes, I did try first M365_downG, later M365_freedom, but only M365_PATCHED seems to do the trick when using the hand-customized firmware bin's.
Perhaps you could update the link to the app at the bottom of https://m365.botox.bz/ .. Thanks again and greetings!

closed 2018-05-23 by waldemar-szostak