BotoX/xiaomi-m365-firmware-patcher #8: Motor power constant - values

oldtrapper, 2018-07-08:
Hi BotoX,

nice Job. I really enjoy it ... pypl donation will follow.

Question for today: Is ther a reason for those odd numbers in the power conastant like 40.165?

What is wrong with 40.000?

By the way, I am 106 kg and max speed reached wiht 40.165 is 28 km/h ... so far ok. Battery temperature after 12 km is 48° (outside air temp about 30°) while battery lasts 10% longer than with stock values and 24 km/h max. Battery lasts even longer than with stock eco mode.

Any experience with battery temerature? max recommended value?

Cheers - Oldtrapper

BotoX, 2018-07-09:
Power constants can be any number, not sure where the odd numbers come from - it's just what the previous custom firmwares chose. They probably took the original value and multiplied it by some percentage and got 40165, etc.
I think the value is the PWM duty cycle, 0 = 100% and 65535 = 0%, so 50% would be 32768 for example.

Battery temperature should not exceed 60°C, 40°C is the perfect temperature for performance, 25°C best balance between performance and battery life.
Max speed depends on the battery voltage, a full battery has 41.5V and can reach ~30km/h, as the battery is depleted the voltage drops and so does the max speed.

allstone, 2018-07-10:
btw with 42V charged battery I can achieve 34.5kmh max

vzoltan, 2018-07-11:
my battery is never more than 41,17x when the charges turns to green.
how did yours charge up to 42V and Botox, yours to 41.5V ?

my max speed with my 88kgs is 29km/h  (if its a free ride -wheel in the air- its max 32km/h)

oldtrapper, 2018-07-11:
I don't care about exact voltage. Just charging til LED turns green. Temperature is more relevant for max speed I suppose. Today it is pretty cold outside and I reached 30.1 k's (according to my garmin GPS Unit) and even on my way back home with about 50 % remaining capacity still 29 k.

BotoX, 2018-07-11:
If you leave the charger plugged in for a while after it turns green it should reach 41.5V, maybe your charger is a little off due to tolerances.
But anyways those 0.5V are maybe 3% difference so it doesn't matter

allstone, 2018-07-11:
I use additional external battery. And charge both together. That is why they charge to 42V. Because internal controller/bms wont let it charge past 41.5V. Reason is that at 42V electronic brakes stop functionic in order not to overcharge the battery. That is why they do not charge to 
42V. Its bad for li-on battery. Best is 41V, thus the green led at ~41V.

vzoltan, 2018-07-11:
Thank you guys for the reply.

Today the mi home app showed max speed of 32,5km/h
Gps shows 30,63km/h

Today was the first time i used mi home app during driving it
Also today i ised the new option of the firmware toolkit 

"Remove hard speed limit of 35km/h"

Before this i just simple set the max speed to 36.

@allstone: Where should i ask you about the external battery option, cuz i just ordered mine yesterday and i dont wanna be off topic in this thread. Thx