BotoX/xiaomi-m365-firmware-patcher #10: Current minimum value on which to start charging mode

allstone, 2018-07-10:
Can you find and introduce minimum amperage value (I think now its 0.1A or lower, at least at 0.01A or 0.02A) when scooter puts itself into charging mode? I mean if charging is lower than 0.1A scooter does not go into charging mode and you can ride. But if its more than 0.1A then it puts into charging mode and you only can turn it off and on back again to ride. This is useful to charge scooter from external battery while riding :)

allstone, 2018-07-10:
updated the value - its -0.05A when the charging mode is on.

BotoX, 2018-07-10:
This is set by the BMS in the BMS struct 16bit value 0x60-0x61: bitflag 0x40 = charging
The ESC firmware checks this bitflag and turns on the brake light effect and disables driving, I look into patching the disable driving part.
I was talking about this today in my telegram chat: https://t.me/XiaomiM365Hacking

allstone, 2018-07-11:
Cool, would be nice to have a flag to disable it ;) I have joined the telegram group as well!