BotoX/xiaomi-m365-firmware-patcher #12: Disable acceleration past a speed point (Motor braking when going downhill)

RedWavve, 2018-07-12:
Hello,
First, thanks for the AMAZING work, I'm flashing my M365 almost everyday trying to solve issues and get the best speed available.

TL:DR > Do you think there's anything that can be done, like disabling the KERS completely, or looking at some flags which would be able to disable acceleration past 28km/h for example

Details:
- Empty, the max speed available is 31.9km/h, whatever my settings are (1000W mode, I got hills and some weight ^^).

However, I'm still having an issue : when going downhill in cruise control or without cruise control, the motor is still running, and that "acceleration" is... producing some braking!
- Kers, Max speed, Eco set at 45
- Power at 25787
- Russian algorithm (I don't have a sinusoidal accelaration with it)
- Hard limit off

I tried the new "remove hard limit", but all it does is "lie" to the driver and divide the speed by 2 so the M365 continues to accelerate, which is useless unless you have a 60V battery. 
- The only solution is to freewheel as soon as I reach the top of the hill by kicking the lever (cruise mode) or just realeasing it, but I bet I'm doing it far worse that a driver could.

Thank you so much ;-)

BotoX, 2018-07-13:
The hard limit patch doesn't lie about anything or divide by 2. It just removes some if speed < 35 check.
And I think the only way to get more downhill speed is to disable the engine and go into free roll mode. This is due to the operation of the motor and the battery voltage limiting the max. speed.

RedWavve, 2018-07-13:
Thanks for the quick answer ;-)
MiHome is reporting 17.2 km/h when that patch is enabled. I'll post a screenshot ASAP.
If anybody could weight in on that and confirm I'm not the only one?

BotoX, 2018-07-13:
That's because the max possible speed that can be display is 2^15 m/h which is approx 32km/h. It's shitty chinese programming, signed 16bit int and sending in m/h when internally the speed is only 345 * km/h accurate...

RedWavve, 2018-07-18:
Hi, is anyone having the same issue?
It seems to be a bit less visible on 1.4.0 than in 1.3.8 for now, but since we still haven't got the changelog from Xiaomi... :-(

rywko, 2018-09-20:
I observed the same when I used FW 1.3.8. Even after releasing the throttle I could still hear noise produced by motor even with KERS disabled.
That's the main reason why I switched to 1.4.0. Once you release the throttle (with KERS disabled) scooter will just roll free with no drag or noise.
I don't recommend turning the scooter off while coasting. Even with my rear brake set properly (wheel skids on full brake), front wheel provides more braking power.

LGM33, 2018-09-21:
> I observed the same when I used FW 1.3.8. Even after releasing the throttle I could still hear noise produced by motor even with KERS disabled.
> That's the main reason why I switched to 1.4.0. Once [...]

I have tried 1.4.0 and it does turn off the motor while coasting.
However, the motor will jerk when restarted. Referenced in issue #11.

This jerkiness happens on brake lever regen and throttle, whenever the motor is reactivated.
This problem seems to be exacerbated by boosted Motor Power Constants.

I have 40165 Power Constant and braking would pulse -7.0 Amp regen current jerking the motor (regen current falls to normal -3A after initial jerk).

What settings are you using? Default Power and Russian Throttle?

I tried Default Throttle with FW 1.4.0. and jerkiness was unrideable.

rywko, 2018-09-26:
I configured Power Constant to 40000. Since I ride mainly on full speed cruise control I've left default throttle alg (there are some issues with Russian alg & cruise control)
Maybe I just got used to it. Actually I sometimes use throttle to slow down on purpose since E-brake kicks even harder

oasbjornsen, 2018-11-01:
Have the same issue. Anyone resolved this?