BotoX/xiaomi-m365-firmware-patcher #15: Option "Remove hard speed limit of 35km/h" does nothing

inaugurator, 2018-07-22:
Hello,
I set the option "Maximum speed in Normal mode" to 40 km/h and option "Remove hard speed limit of 35km/h" in the firmware version 1.3.8 and 1.4.0. But after that the maximum speed does not exceed 30 km/h, even when I'm driving from the hill.
Is there a solution to this problem?
Thank you.

BotoX, 2018-07-22:
Physics, the voltage of the battery determines Vmax. Releasing the throttle while going downhill could help. Also use GPS speed.

inaugurator, 2018-07-22:
I used GPS!

BotoX, 2018-07-23:
Pretty much, maybe not downhill but idk..
I added it because I was having stuttering issues with my 12S battery.

inaugurator, 2018-07-23:
Thanks for considering my request.

closed 2018-07-23 by inaugurator

roobre, 2018-08-17:
Hi,
Sorry for reviving a closed issue, just wanted to ask something related.

> Physics, the voltage of the battery determines Vmax.

Is this actually true? Being a BLDC motor, battery voltage should only determine the max torque/power (and not speed directly), since the rotation of the magnetic field is electronically controlled by the hall sensors to have it always pushing the rotor. Transistors/sensors should be fast enough to provide near the thousand of rpm (35km/h fall into the range of hundreds) if there is no resistant torque, which is something that driving on a flat surface (or even downhill) shouldn't generate.

Just wanted to hear your thoughts about this and double check if this issue is completely unrelated to https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/14

BotoX, 2018-08-17:
With field weakening that's absolutely possible, however during normal FOC operation the voltage determines the maximum RPM. I don't really know the details of this however, but it's the same on e-bikes and etc. that you need a higher voltage system to reach higher speeds.

roobre, 2018-08-20:
Perhaps you're right, electrical stuff is a bit off from my field of expertise, so I won't argue about it. However with the issue I'm experiencing, which is similar to this one and #14, I can feel how the engine suddenly stops accelerating, and after a few secs it starts again. Almost like it triggered a hardcoded barrier and the power were abruptly cut.

Just wanted to comment it because probably we can extend #14 to normal mode too when the russian throttle algorithm is enabled, and perhaps it was related to this one too.

Anyway, thank you for your nice work and please tell me if I can be of some help.