BotoX/xiaomi-m365-firmware-patcher #16: Firmware block in 1.4.1

SheikhEdward, 2018-08-08:
Is there any way to remove this block ? Was to hasty to press yes on the install firmware request

Kind Regards,

Made a lot of .BIN files to try your fine utility. Did go faster than 35 km/h on a flat surface and
I am way over 60 kg

Edward

BotoX, 2018-08-09:
Yes, using an ST-Link and this guide:
https://electro.club/f/48886

No other way is known so far.

SheikhEdward, 2018-08-10:
😁😎 I have to ask our Russian friends to translate it, my understanding of the language is a but rusty
Thanks

MrSnowball-dev, 2018-08-10:
There's a much simpler way (though you still need ST-Link): https://electro.club/f/49764

There you need an ST-Link CLI and your scooter's serial number. Input serial number and path to CLI to that little program and click the button.

This will wipe your mileage, FYI.

jolie16, 2018-08-12:
I looked at firmware 1.4.1 in which the decryption functions are already integrated. Here's what I found out:
The encryption algorithm is similar to [TEA (Tiny Encryption Algorithm)](https://en.wikipedia.org/wiki/Tiny_Encryption_Algorithm).
Keys are:
```
private_key.key_0 = 0xB21C80FE;
private_key.key_1 = 0xA641EFD1;
private_key.key_2 = 0xF53117A4;
private_key.key_3 = 0xF02468A0;
```
Encryption code and key round changing function are on screenshots. Also archive with disassembled DRV firmware 1.4.1 in IDA PRO is attached.

Maybe you can test that and add option "Encrypt firmware" to constructor.

PS.: Sorry for my english.

[image]
[image]
[image]

CharlieSpringer007, 2018-08-15:
Hi, I ordered a ST-link but not sure how and what to connect the ST-link to on my m365 scooter. Could some please let me know how to connect it? Thanks

jbfuzier, 2018-08-19:
There is a way to downgrade now :

https://gyronews.com/downgrade-m365-1-4-1-cest-possible-2142/

Vagos696, 2018-12-02:
Confirmed that this bin downgrades to 1.4.0 and then you can apply firmwares freely.
You have use downG patched.

SheikhEdward, 2018-12-02:
Thanks, did it some while ago when the total panic was there. After the Spanish .BIN flashed to 1.40all was fine
M365 4ever
Regards,Edward

MyTheValentinus, 2018-12-23:
Hello,
My M365 is brick, when i check the firmware version with the android app patched, i have a strange number: -10201
And i am unable to connect on Mi Home app or Ninebot too
I will buy a AT-link to flash the board, but how to get my serial number and model number ? Where is ?
Thanks

SheikhEdward, 2018-12-23:
Hi Valentin, my name is Edward. I am afraid I am notfamiliar with a bricked M365.
Just out of curiousity, how did you do that ?
I normally use the Bluetooth connection and haveone app on Android to do the official .BIN filesand another app to do the Botox.cz .BIN files.
With these apps, you can not connect anymore ?
Regard,Edward

MyTheValentinus, 2018-12-23:
Connection Bluethoot with the com.m365downgrade-v7_PATCHED.apk ok, but when i check the version firmware is mark as -10201, and when i try to flash any .BIN, error at 4% retry...
I buy this scooter on internet, and i think the seller try to flash custom firmware, but failed.

SheikhEdward, 2018-12-23:
> Yeah, battery is okay, i have charged the battery more than 12 hours. Bluethooth connection ok, but unable to check version i have a -10021 return

Ok, also with the other app ? No luck in flashingwith that app. Never had that error actually

acoryw, 2019-02-20:
Any word on that version -10021 sent retry error?

BotoX, 2019-02-20:
-10021 error comes from a fake/clone BLE.

omgbox, 2019-03-15:
> -10021 error comes from a fake/clone BLE.

would original m365 board bypass this problem ?  the clone does not appear to have Bluetooth to flash firmware, when flashing it stops  at 4%........ and repeats until it disconnects. 

not able to connect with mi home app, but can connect with another app on IPhone. to check battery .

current firmware 1.5.1, does not allow other apps to connect to Bluetooth. on the apple stores