BotoX/xiaomi-m365-firmware-patcher #18: different power for eco and normal mode

ataomega, 2018-08-18:
Hi, At first thanks for your great tool! well done!

Not sure if it's the right place to ask this, Is it possible to have different power levels for normal and eco mode, for example I'd like to have motor power of 25787 in normal mode and 51575 ( stock ) in the eco mode. How can I achieve that?

Thanks!

BotoX, 2018-08-18:
the motor power constant is the maximum duty cycle, it doesn't differentiate between normal and eco
but eco has lower acceleration by default in the throttle control algorithm so it will result in lower power than normal mode

BotoX, 2018-08-18:
It affects the brake, it probably does affect KERS a little bit too.

BotoX, 2018-08-18:
And that constant is just the "maximum power" not continuous. So you get more power when accelerating and climbing hills, speed, wind, drag, weight, heat, etc. influence range.

Nickduino, 2018-09-05:
I was wondering the same myself.

If the duty cycle is the same for both mode, why do we see custom firmwares advertising 400 W in Eco and 800-1000 W in Normal, then?