BotoX/xiaomi-m365-firmware-patcher #19: Battery modification - 10s5p

beckos, 2018-08-18:
Hi! I plan on modifying the battery from 10s3p to 10s5p by adding two more LG M26 cells in parallel for each of the ten groups of three paralleled cels. The resulted capacity of 13 Ah would be managed by the same BMS and charging controller. I expect that declared capacity of 7800mAh be displayed the same in any app...

My question and proposal is: could this declared capacity be modified so that 100% charge and estimated range would be displayed correctly for the newly configured pack of 13Ah?

BotoX, 2018-08-19:
It'll be fine since the BMS calculates the remaining capacity based on voltage (which is grossly inaccurate)
And I believe you can also override the capacity in the BMS memory with a simple write command over serial/bluetooth.
You could also sell your original battery and build a better one with better cells and a proper BMS like I did: https://cloud.botox.bz/s/cMifL7kWyMwLA8n
BMS: https://www.lithiumbatterypcb.com/product/13s-48v-li-ion-battery-pcb-board-54-6v-lithium-bms-with-60a-discharge-current-for-electric-motorcycle-and-e-scooter-protection-2-2-3-2-2-2-2-2/
Open source firmware compatible with m365 can be found here: https://github.com/BotoX/xiaomi-m365-compatible-bms

beckos, 2018-08-19:
Hi!

Wow! Thank you so much for sharing this info! 

Indeed that is a huge energy core upgrade! 

It is a bit costly at the moment for me, and may make the scooter too heavy, as I have to carry it 3 floors up & down. I wanted to just add a bit more energy for a day long ride.

If I may ask: 

- Does the controller and charging inside the original mainboard can support 12s4p voltages by default? Or did you changed that as well? 
-How do you charge the custom pack? At what rate? Can it be fast charged somehow? 
- And the second battery, you connect it in parallel with the main one? But where do you keep it on the scooter? 

Many questions... thank you for sharing!

C. Burlan (mobile)

BotoX, 2018-08-19:
I'm using the original controller with no modifications, no problems with the higher voltage.
I charge it with a Mean Well HLG-240H-48A which can charge with up to 5A (240W), way faster than the original charger.

I don't have a second battery, but yeah it'd be connected in parallel to the main one and people put a bag on the 45° pipe in the front and keep the external battery there.

beckos, 2018-08-19:
Interesting, because this is just a CC+CV LED driver! I can see that Voltage and Current can be adjusted. The original onboard controller then regulates the charging current according to what data it receives from the BMS? I don't have a datasheet for LG M26 cells to see what is the maximum charging current supported, or charging current chart. If external supply is able to supply more current the controller will draw to the maximum supplied and then it would regulate the charging current curve? The original BMS must be reprogrammed to higher charging currents?

BotoX, 2018-08-19:
There is no charging circuit on the scooter.
The charger does all of the charging stuff :D
And yeah all you need for lithium batteries is CC/CV. The current goes straight into the batteries. I suppose there might be some overcurrent protection on the BMS.
And about the max. charge current, for lithium 99% of the time 0.5 * C (capacity) is a good value.

BotoX, 2018-08-19:
The BMS can not control the current, it can only shut the battery off completely.

Please look up what CC / CV means.

BotoX, 2018-08-20:
That something is the CC/CV charger or rather the battery itself as it gets full.

alexeypetrenko, 2018-10-26:
Hi, Sorry for bringing up an old thread.

@BotoX , Your custom battery project is amazing! I was thinking about making a custom battery for my m365 as well. Could you give a bit more information?

 - How does BMS interface with a controller? Is it just a matter of connecting UART of a BMS directly to a controller, or is there any other circuitry in between?
 - Do I have to patch the main controller to get it to work with this self-made battery?
 - Rear LED is not connected in this case?

Hermit12, 2018-10-27:
Hi, I also have some questions,
I got a 13s battery from my old scooter and try to fit it in the m365. This should work by putting the controller in the back and the battery in the front. However I would need to change the bms so that the m365 controller is able to recognize it, right? Will 48V work with the stock controller and do you know how much amps the controller can take before it melts?
I build some scooters with a lot of China stuff and they work great, but the look is always selfmade and not so fancy so I try to get the same performance in this good looking xiaomi scooter. 

Regards and thanks for your help

BotoX, 2018-10-29:
> * How does BMS interface with a controller? Is it just a matter of connecting UART of a BMS directly to a controller, or is there any other circuitry in between?

Yes, the scooter communicates with the BMS via simple UART. Just connect the BMS to the 3pin on the scooter that says R T L. R = Receive, T = Transmit, L = Light.

> * Do I have to patch the main controller to get it to work with this self-made battery?

No, except if you want to use a higher voltage, then you'll have to use the higher voltage patch for the firmware.

> * Rear LED is not connected in this case?

The 3rd pin on the RTL connector is the positive of the brake light. Just make your own cable and connect the light positive to that and the negative to ground.

> I got a 13s battery from my old scooter and try to fit it in the m365. This should work by putting the controller in the back and the battery in the front. However I would need to change the bms so that the m365

I wouldn't put the controller in the back, you'll have to redo all the wiring....

> However I would need to change the bms so that the m365 controller is able to recognize it, right?

Yeah, see my first comment in this thread.

> Will 48V work with the stock controller and do you know how much amps the controller can take before it melts?

Some ICs on the controller will be destroyed if the bus voltages goes above 63 volts!
When you brake the motor generates power and the bus voltage goes up by a few volts.
13S is 4.2*13 = 54.6V on a full battery, when you hit the brakes hard you'll probably see up to around 58V for a short time. This should be fine still.

alexeypetrenko, 2018-10-29:
Wow, that is amazing!

Couple more questions:
 - You are using 12s battery and your code sets short circuit protection current to 60 A. Does it mean that you bought 60A capable bms?
The thing is, I do not see `12S li-ion 60A` option in a [store](https://www.lithiumbatterypcb.com/product/13s-48v-li-ion-battery-pcb-board-54-6v-lithium-bms-with-60a-discharge-current-for-electric-motorcycle-and-e-scooter-protection-2-2-3-2-2-2-2-2/). Is it out of stock, or have you bought something like `13S li-ion 60A` / `14S li-ion 60A` and simply connected less wires? Is it even supported?
 -  Do you think it would be fine to put `Samsung INR18650-35E` batteries in 12s4p pack? A single `INR18650-35E` cell has maximum constant discharge current of 8A. It means that the pack will have 32A maximum constant discharge current. From what I cold find M365 uses `LGGBM2618650`, which seem to have 10A discharge current, or 30A for pack.

BotoX, 2018-10-29:
> * You are using 12s battery and your code sets short circuit protection current to 60 A. Does it mean that you bought 60A capable bms?

I bought the 30A version because it only has mosfets on one side, the bigger versions would be too thick.

> * The thing is, I do not see `12S li-ion 60A` option in a [...]

I got the 12S 30A one, it does say lifepo4 but that's just a software setting.

> * Do you think it would be fine to put `Samsung INR18650-35E` batteries in 12s4p pack? A single `INR18650-35E` cell has maximum constant discharge current of 8A. It means that the pack will have 32A [...]

These ratings are max. continuous current, you're unlikely to be using 30A all the time. Your engine would go up in flames before the batteries would care. I used NCR18650B in 12S4P configuration and they aren't getting warm at all. So yes, INR18650-35E is perfectly suitable for this job.

pitslz, 2018-11-25:
@BotoX You are amazing for sharing this battery updrade!!!
How and where do you upload BMS communication firmware?

kzkz22, 2018-11-27:
> It'll be fine since the BMS calculates the remaining capacity based on voltage (which is grossly inaccurate)
> And I believe you can also override the capacity in the BMS memory with a simple write [...]

Dear BotoX,

Your original cloud link is broken. Can you share its content somehow?

Can you give some tutorial how to modify, how to flash the custom BMS to work with original M365 controller?

Thank you in advance!

Regards, 
Zoltan

BotoX, 2018-11-27:
> Dear BotoX,
>
> Your original cloud link is broken. Can you share its content somehow?
>
> Can you give some tutorial how to modify, how to flash the custom BMS to work with original M365 controller? [...]

Weird that the link changed: https://cloud.botox.bz/s/94drnBJfjacBDnr

I'll try to write a tutorial for flashing the BMS this week, kinda busy and writing stuff isn't really something I enjoy :p

FlakesIce, 2018-11-27:
BotoX, how'd you end up closing the scooter back up? My battery is just like yours but I need around 5mm more of clearance. I'm thinking of making a small 3d printed frame and elevating the whole cover by that lenght. What was your solution?

BotoX, 2018-11-28:
> BotoX, how'd you end up closing the scooter back up? My battery is just like yours but I need around 5mm more of clearance. I'm thinking of making a small 3d printed frame and elevating the whole cover by that lenght. What was your solution?

I took some 5mm thick rubber sheet (6mm would've been better, used some rubber foam in addition), cut it with a utility knife and used that in between the scooter and the bottom for the extra height and as a waterproof seal :+1: 

Also I finally got around writing that README: https://github.com/BotoX/xiaomi-m365-compatible-bms

kzkz22, 2018-11-28:
> > BotoX, how'd you end up closing the scooter back up? My battery is just like yours but I need around 5mm more of clearance. I'm thinking of making a small 3d printed frame and elevating the whole cover by that lenght. What was your solution?
> 
> I took some 5mm thick rubber sheet (6mm would've been better, used some rubber foam in addition), cut it with a utility knife and used that in between the scooter and the bottom for the extra height and as a waterproof seal 👍
> 
> Also I finally got around writing that README: https://github.com/BotoX/xiaomi-m365-compatible-bms

Thanks for your detailed instructions!

I found two bottom spacers at thingiverse:
https://www.thingiverse.com/thing:3188558
https://www.thingiverse.com/thing:3192237

pitslz, 2018-11-28:
You are in GOD mode. Amazing README! I'll share my live experience when the my parts arrive.

BotoX, 2018-11-28:
Here's my Fusion360 3D model of the bottom cover: https://a360.co/2RkB9GJ
You can also export it to any other format you like.
I've used that to laser cut a piece of wood and used that as a template for cutting the rubber spacer.

kzkz22, 2018-11-30:
My plan is to convert the original 10s3 pack to 10s4p like this way. It is possible without modifying any other component like BMS, controller board etc.?

I think it is cheaper and easyer method than building a new pack with custom BMS. 

[image]
[image]

BotoX, 2018-12-01:
Sure you can do that, works just fine with the original BMS too.
But why not just add an external battery in parallel?
Or sell your original battery and build a big one if you have the tools?
This is not a good solution in my eyes.

kzkz22, 2018-12-01:
> Sure you can do that, works just fine with the original BMS too.
THX!

> But why not just add an external battery in parallel?
If I add external battery in parallel than I won't have BMS information of that battery pack.

> Or sell your original battery and build a big one if you have the tools?
I haven't got spot welder yet, but I'm planning to buy one.

> This is not a good solution in my eyes.
Why? :) I need only a little more capacity, and I get about 30% plus capacity with this method which is enough for me.

BotoX, 2018-12-01:
> If I add external battery in parallel than I won't have BMS information of that battery pack.
The original BMS calculates the battery capacity from the voltage (which is really inaccurate), so there are no issue with adding an external battery.

jbfuzier, 2019-01-21:
Thanks, that seems like a bad thing to do :(
There must be some kind of electronic roughly controlling how much power is send (low, medium or high settings) ?

What happens when the voltage from the kers is bellow the one of the battery ? I guess that would not allow to get a current flowing to the battery, but then where will the energy go ?

Finally, which settings are you using for the kers on your 12s ?

miloszlip, 2019-04-05:
@jbfuzier I have questions to You. Where have You place this 2s addtional battery? What's is the impression of the ride?

jbfuzier, 2019-04-05:
I put it in a bag, I will take a picture to show you. You get much more power and acceleration with it, it is great to climb hills.
You get the same behavior as with a 12s pack, physically speaking it is the same voltage, you will be able to draw the same amount of current...

jbfuzier, 2019-04-08:
[image]

miloszlip, 2019-04-09:
Thats great. I am wondering if there is possibility to charge this additional pack with only one standard charger. I thinking of dc-dc step down converter to achieve that.

RedDevils99, 2019-04-12:
"I'm using the original controller with no modifications, no problems with the higher voltage.
I charge it with a Mean Well HLG-240H-48A which can charge with up to 5A (240W), way faster than the original charger."

BotoX, thanks for all your work to make the Xioami a more fun experience. I can't flash my firmware because I already got the 1.5.1 update. I hope someone cracks that block soon. I have a noob question:

I have a 48v 9ah (their words) BH Easy Motion battery handy, since my Nitro City got stolen, and i have it's charger. I have a Xiaomi M365 with the two fuses. I have bravery and very basic soldering skills.

Battery storage issues aside, could I connect the 48v 9ah battery to the stock controller and ride off without messing up the battery? How about the controller? 

Would it make any difference performance-wise without any changes to anything else?

I was always frustrated by the 9ah thing on my bike because it made my range short. But it might be perfect if the direct wiring works.

Thanks again for sharing your tools and knowledge.

BotoX, 2019-04-13:
48V = 13S, that's too much for the scooter, 12S max.
And you'd need a compatible BMS or emulator.

jbfuzier, 2019-04-14:
@miloszlip

I would not charge both pack together as they have 2 independent bms, if you want to charge both together the best course of action is to build a 12s pack with botoX's bms.

Regarding the step down, I would say it is a bad idea, chargeurs for liion cells are not dumb power supply, they follow charging step rules delivering constant current / constant voltage when required, I think a step down would break that (not 100% sure).

RedDevils99, 2019-04-14:
"48V = 13S, that's too much for the scooter, 12S max.
And you'd need a compatible BMS or emulator."

So earlier in this thread, you meant that you used a 48v 12s when you said "I'm using the original controller with no modifications, no problems with the higher voltage," you had replaced the bms but were using the original controller? Or how did you get 48v 12s not 13s?

The battery has it's own bms, could I use that with a new controller?

 I'm sure these questions are basic, thank you for your reply.

pedala, 2019-04-15:
> @miloszlip
>
> I would not charge both pack together as they have 2 independent bms, if you want to charge both together the best course of action is to build a 12s pack with botoX's bms.
>
> Regarding [...]

I run an external 10S4P in paralel (on the load side) with the internal stock battery. I charge both at the same time with via the original charge port. Each BMS does it's own balancing and disconnecting in case of overCurrent/overVoltage/underDischarge. With the stock charger (41.8V, 2A) it takes ~12h for a full charge. Some 60-70% of this current goes into the stock battery, as it has lower impedance. This is also evident while riding - the current draw is higher than half on the internal, so after you stop, the external battery charges the internal with a small current (0.3 ... 1.5A). 
I just built my custom charger - 41.8V just to be safe, current limited at 6A. I expect max 70% to go to the internal battery, so 3.5 ... 4.5A max on the internal one. (3.75A is recommended as per Datasheet)
Even if one battery gets disconnected (BMS, user, vibrations etc) the full 6A will not damage the remaining battery.

> 
> Regarding the step down, I would say it is a bad idea, chargeurs for liion cells are not dumb power supply, they follow charging step rules delivering constant current / constant voltage when required, I think a step down would break that (not 100% sure).

Most chargers are just CC/CV, no extra smart parts inside (max charge timers, mAh counter, end-of-charge current threshold etc). Agreed, you can't just a use any 42V supply, even limited to X Amps, it must be one that goes into CC mode, not just disconnect the output when over current is reached.

BotoX, 2019-04-15:
Here's all the info on the 12S battery / BMS: https://github.com/BotoX/xiaomi-m365-compatible-bms
I didn't modify the original motor controller, ofc the original battery (including it's BMS) is not used anymore and was sold

edreyyo, 2019-04-24:
> And yeah all you need for lithium batteries is CC/CV. The current goes straight into the batteries. I suppose there might be some overcurrent protection on the BMS.

A little off-topic @BotoX, but please can you confirm whether the power brick included with the m365 is a CC/CV li-ion charger or just a regular PSU? Why I ask is because mine just says "power adapter" on the unit. There is no mention of li-ion charger.

Perhaps CC/CV is achieved by some circuitry within the scooter itself?

Thanks

BotoX, 2019-04-24:
The Xiaomi charging brick is a CC / CV power supply.
The BMS can only turn the input / output of the battery ON or OFF.

edreyyo, 2019-04-24:
> The Xiaomi charging brick is a CC / CV power supply.
> The BMS can only turn the input / output of the battery ON or OFF.

Excellent, this means I can also use it to charger my 10s4p electric skateboard pack. Thanks!

pedala, 2019-04-25:
Followup on my external 10s4p battery: 
Stock BMS disconnects above 2.4A, so I charge both batteries in paralel at 4...6A max. I am using a DPS5015 power supply with a 60V input

ricardojd, 2019-07-12:
Can someone tell me what I need to do to put my scooter with more speed. How can had more batteries to get more voltage. Someone have some project with list of material I need to increase voltage without burn BMS. Thanks.

ricardojd, 2019-07-25:
Hello. Someone know if this battery fit on Xiaomi m365 and get more speed? And I need to change anything like BMS... Or it's only connect.
https://s.click.aliexpress.com/e/4KJsMLe

rstige, 2019-08-08:
Hi. The bms listed is not available on AliExpress any more. Will this one work as well.

https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.aliexpress.com/i/33061290001.html&ved=2ahUKEwiFn_iKovTjAhUsz6YKHTNLC70QFjAAegQIARAB&usg=AOvVaw2SQVdUj8YGmWE4bqiYxYbs

BotoX, 2019-08-09:
You can, pay attention to the datasheet however: http://www.ti.com/lit/ds/symlink/bq76940.pdf
Table 26. Cell Connections for bq76940
on Page 42

rstige, 2019-08-09:
> You can, pay attention to the datasheet however: http://www.ti.com/lit/ds/symlink/bq76940.pdf
> Table 26. Cell Connections for bq76940
> on Page 42

Thanks for you help. Very much appreciated.

Killerkickan, 2019-09-01:
I have put an extra battery on my M365 with 7000mAh and it´s working great have ride over 350 km with it and have no problems with the BMS or anything else. So know i have 14 800mAh and can ride about 40-50 km depending of how fast i go, but with a average speed of 25 and sometimes going over 30km/h it holds 45  km easy. I started with the solution with a Y connector but then the DASH DIsplay dont worked so I solder the cables direct to the original battery and now everything works perfect.

[image]
[image]

edreyyo, 2019-12-30:
This is epic @Killerkickan, thanks for sharing.

KrzysztofHajdamowicz, 2020-05-31:
Have anyone tried to merge 3 battery packs in parallel? One stock battery and two range extenders  through Y-shaped cable?

C0br4, 2020-07-21:
> Have anyone tried to merge 3 battery packs in parallel? One stock battery and two range extenders through Y-shaped cable?

You can parallel as many batteries as you want or can physically fit on the scooter as long as their cell count and voltage match

yomuevans, 2020-09-16:
> I have put an extra battery on my M365 with 7000mAh and it´s working great have ride over 350 km with it and have no problems with the BMS or anything else. So know i have 14 800mAh and can ride about 40-50 km depending of how fast i go, but with a average speed of 25km/h and sometimes going over 30km/h it holds 45 km easy. I started with the solution with a Y connector but then the DASH DIsplay dont worked so I solder the cables direct to the original battery and now everything works perfect.
> The first pic is when i solder the extra battery on the main.
> The second pic is when i did a Y connector that worked great until I put the DASH Display on. The Dash Display have a setting that makes the display only say Charging when I used the Y connector but if you don´t have any Displat on your M365 i recommend this before solder cables on main battery. And if anything happens you just take it of and return it were you bought it and the can´t see that you have done anything to the scooter.
> 
> [image]
> 
> [image]

Hi, were you able to charge both batteries at the same time?

Killerkickan, 2020-09-16:
Hello, yes no problems to charge both battery. But I have solder the cables direct from the extra battery to the main battery and with a fuse between the extra battery and original. I also test the voltage after  charging once or twice a month. But have driven almost 700 km now and they still after charge 41,7-41,8 on both battery. They ha e always the same voltage but sometimes it's ,7 and sometimes ,8.

Mvh Anders Malmström

Tessinsväg 10 B
217 58 Malmö
0763-22 23 25

[Solder-on-original-battery]<https://user-images.githubusercontent.com/53076901/64073851-dd892900-cca3-11e9-9c50-21ab9968d287.jpg>

[Y-connector]<https://user-images.githubusercontent.com/53076901/64073691-99952480-cca1-11e9-985b-b01ba9c49cad.jpg>

yomuevans, 2020-09-26:
Thanks, 1 last question, my questions might be silly kindly Forgive me, what do you mean you soldered directly to the main battery? I understand there's an xt30 plug to the controller, was that the point of soldering or you made a y connection yourself?

Killerkickan, 2020-09-30:
Sorry for later answer, it´s was in the Spam and I had just sent a 500 
mail vote an got about 150 mail in the Spam  😊.
I´m not sure what you mean but I took a pair new cables to the extra 
battery and solder them on the original battery instead of th Y 
connection!
Because if I use the Y connector my Dash365 Display didn't worked. So I 
have cables from the olf battery to the Extra battery with a Fuse on the 
cable. So you can say that my control box think I have one battery 
instead of 2! But you should know what you are doing if you are going to 
solder on the original battery, can easy be a fire and battery can even 
explode if you have the solder iron to near and long against the battery

Med vänliga hälsningar Anders Malmström
Tessinsväg 10 B
217 58 Malmö, Sverige
+46 763-22 23 25

Kopia: "Killerkickan" <killerkickan@gmail.com>; "Mention" 
Skickat: 2020-09-26 06:35:41
10s5p (#19)

yomuevans, 2020-09-30:
Thanks this was helpful, I think ill do the same, am still worried about the 2nd not a data cable line

kilabots, 2021-04-22:
hello. i plan to monitor the voltage percentage and connect it to esp32. do you know how to do that?