BotoX/xiaomi-m365-firmware-patcher #22: CFW version spoofing

1upus, 2018-09-10:
Is anybody knows adress, where i can patch fw version, like 1.3.8->9.9.9?

BotoX, 2018-09-10:
Version is saved as hexadecimal -> search for 0x138, can provide details if needed.
Wanted to make a signature out of it but all of the code around it changes slightly on every update.
I suppose you are annoyed by the update notification in the mi home app? Use a 3rd party app like m365 tools or m365 HUD, much better anyways.

1upus, 2018-09-11:
Thx. Any crc or another hash checks? I can just patch fw version?
And yes this all about update notification in MiHome and Ninebot apps.

BotoX, 2018-09-11:
You can just change the number and that's what the scooter will report to the app.
The app will then not bother you anymore with updates.

But as said before, I strongly recommend ditching the official apps and using this instead:
https://play.google.com/store/apps/details?id=app.peretti.m365tools

rshev, 2019-01-06:
@BotoX could you please take a look at these offsets again? (the reason is similar, but there is no  m365tools, m365hud for iOS)
I can find hex 01 38 in 1.38 firmware, but no 01 34 in 1.34 firmware, meaning that it's either incorrect or not little endian. Searching for 38 01 gives too many matches => would not like to guess in this case 😄

toegema, 2019-12-12:
> You can just change the number and that's what the scooter will report to the app.
> The app will then not bother you anymore with updates.
> 
> But as said before, I strongly recommend ditching the official apps and using this instead:
> https://play.google.com/store/apps/details?id=app.peretti.m365tools

Hi, Botox. 
The app you mention is a very nice app. I only use that one now. BUT, it would still be nice to be able to make the version updates go away in the original app. The reason for that is that here in Norway, the M365 is illegal because of our 20 km/t speed limit. Even if I change the firmware on these scooters and limit the speed (I do it for all my friends scooters also, since almost all have that model), as long as the user can change the speed back to default, the fix is not legal. Only if the user “cannot” undo the speed limit it will be considered legal. And since the average user does not “hack” firmware, it the firmware update possibility went away in the original app, all would be good. So if this is something you could add without too much hassle it could make the possibility to make these scooters legal in Norway. 😉 Merry Christmas!