BotoX/xiaomi-m365-firmware-patcher #24: Feature request: switch between Eco and Normal mode with ONE click

bibicu123, 2018-09-28:
Hi !

Thank for your great work.

I wonder if it is possible to add a patch that enables us to switch between Normal and Eco mode with one click instead of a double click. And turn on/off the headlight with a double click. 

The reason behind this is:
- In my city  many roads are bad, so I use Eco mode because I have to go slow, and brake often.
- But there are some roads that are good and I switch to Normal mode and max speed on these.

So during a journey I toggle between Eco/Normal modes many times, but I turn on the headlamp only once maybe.

So it would be much more user friendly to use one click for Normal/Eco and double click for headlight.

I know there is a Instant Switch from Normal to Eco patch, but it still needs a double click.

Thank you !

bibicu123, 2018-09-28:
I suppose the code is calling a subroutine based on a button event.

if(single_click) then call toggle_headlight();
if(double_click) then call toggle_normal_eco_mode();

Maybe you can find a way to reverse the calls.

if(single_click) then call toggle_normal_eco_mode();
if(double_click) then call toggle_headlight();

Regards

BotoX, 2019-01-08:
The button code is actually in the BLE firmware  and I'm not planning on patching the BLE firmware.

closed 2019-01-08 by BotoX