BotoX/xiaomi-m365-firmware-patcher #27: Battery Modification 11s3p OR 12s3p

leogiouldasis, 2018-12-03:
Hello,
I am interested in upping only the voltage on my battery for extra kick, without affecting lots of other stuff in the scooter. From my understanding the firmware patcher allows for overriding 42v max to bigger which is goiing to be needed.
What i don't know is:
Do i have to get a different bms for that to work? the 11s3p scenario will add another 3 batteries in series that will hopefully fit inside the original area without modifications and just connecting the 3 in the bms. 
Or i need a new bms, which in that case i may rework the battery from scratch and go for bigger voltage and capacity ( 12s4p ).

Also the scooter is plug and play with different battery/bms setup? I suppose you can't flash the bms afterwards like you used to if my understanding is correct.

BotoX, 2018-12-03:
If you want to go the cheap and easy way then you can just add a 2S3P with the same or better cells in series. Use a cheap 2S BMS for it that can do 30A.
However how do you charge that stuff then, do you charge it together with the main battery (bad, no common BMS) or charge the main battery and the 2S separately (two chargers)?
The original BMS only supports 10S.
If you are up for the task I definitely recommend building a 12S4P pack and selling your original battery. I use a Mean Well HLG-240H-48A for charging it.

leogiouldasis, 2018-12-03:
Thank you very much for the recommendation.
If i go building a 12s4p pack do i need a custom bms firmware to be programmed?

BotoX, 2018-12-04:
Yes of course, read this: https://github.com/BotoX/xiaomi-m365-compatible-bms

leogiouldasis, 2018-12-04:
> Yes of course, read this: https://github.com/BotoX/xiaomi-m365-compatible-bms

Thanks for the input, i think i am not qualified for such a task yet, so i will go the cheap and easy way of adding a 2s3p with 2900cells and charge seperately. I am struggling to find a 30A bms though from europe? Any recommendations?

BotoX, 2018-12-04:
> I am struggling to find a 30A bms though from europe? Any recommendations?

Yes, order from AliExpress ^^

closed 2019-01-08 by BotoX