BotoX/xiaomi-m365-firmware-patcher #30: Remove 50% and less power limit

Baterka, 2019-01-06:
There must be way how to remove this weird function what limiting power and top-speed undee +-50% on battery.
Yea sure, we cannot change physics but 300W on 50% is NOT maximum what batteries can push on this state.
When you drive for example other custom scooter with VESC (and even slower HUB motor), there is almost no change in power delivery until 20% of battery.

Maybe its caused by BMS, is there way to edit BMS fw?

BotoX, 2019-01-08:
It's definitely not caused by the BMS (as that can only turn the battery ON or OFF) and I did not find any clue in the ESC firmware for this (power is limited at <800mAh)
Top speed (max RPM) also depends on battery voltage btw.
I might take a look at it again some day but I'm pretty sure it's just physics.

Baterka, 2019-01-08:
Power is limited <800mAh on what version? Because Its almost 100% perfect 300W limit right after reaching 50%

BotoX, 2019-01-08:
Even when driving uphill or accelerating?
Because driving straight you don't need more than 300W anyways lol.

emileb, 2019-02-07:
Yes this is actually the only function I am looking for in a custom firmware image. There is definitely some code somewhere which is limiting power/speed below 50% which is really annoying.

Baterka, 2019-02-07:
> Even when driving uphill or accelerating?
> Because driving straight you don't need more than 300W anyways lol.

When driving on flat road it is +- 400W in 25km/h (when >50%)

Baterka, 2019-02-07:
> Yes this is actually the only function I am looking for in a custom firmware image. There is definitely some code somewhere which is limiting power/speed below 50% which is really annoying.

Yea... Tell me why I am driving 28km/h on 90% charged battery and taking +-450W and I can't drive 25km/h when I am under 50%, BUT UPHILL under 50% is giving me 700W??? :D This is complete nonsense... Batteries can push more than 700W in >20% when uphill, so why not on flat road?

noctrex, 2019-02-12:
Have you tried with the Russian throttle algorithm enabled? For me this gets the job done. When returning home from work, towards the end I have a small percentage uphill but very long, and it always gets below 50%. With the original throttle algorithm, the scooter actually stops before reaching the end, as it doesn't provide enough power. But with the Russian algorithm, even when the battery is below 50%, I can go till the end of this uphill.

lukasd1, 2019-03-07:
I using the russian algorithm and unfortunatelly is the same like with normal. Maybe the problem is not a limit under 50% but under 20 celc. deg. of battery temperature ?  I'm testing a few days and when my battery has about 18 deg. there is no power , but when is more than 20 problem wasn't exist(or it was smaller)  . I've extra battery pack and temperature of main battery usually is lower as normal.

ryci-us, 2019-04-07:
I think its a physics.
I have upgraded battery to 12s3p with samsung 30q cells and aftermarket BMS with Botox firmware and i don't have such effect. Acceleration is about the same in whole range from 9000 to 2000 mAh, only top speed changes.

miloszlip, 2019-04-15:
Can you provide us some details on this 12s3p battery? Will it fit on the oryginal battery spot? And can You tell me which BMS You are using? What is the top speed?

BotoX, 2019-04-15:
https://github.com/BotoX/xiaomi-m365-compatible-bms
35km/h max realistic
12S3P fits with original lid
12S4P needs 5mm - 5.5mm spacer

miloszlip, 2019-04-17:
Thanks a lot! Do you have some photos of this battery / whole mod?

BotoX, 2019-04-18:
It's in the README if that GitHub I linked: https://cloud.botox.bz/apps/gallery/s/94drnBJfjacBDnr

MatG0, 2019-07-29:
There is some software function which is limiting output power and speed depending on battery level % or voltage
At first 50% it is quite decent and linear. The max speed drops from 31kmh@100% battery to ~28kmh@70%. It is not physics because even when you go downhill, scooter will start brake when you reach 28kmh at 70% battery. There is some speed limiter which is independent on known/available firmware settings and is related to battery level. Scooter will not pass 32kmh, whatever you set and the speed limit even lowers with decreasing battery level...

Depending on "tuned" settings the peak power is ~750W for me (motor constant ~36000) at 100% battery. 
At 65% I am still able to get the 700W peak output at 34,5V.
At 50% I am limited to ~430W or less peak at same voltage. 
It may seem like the firmware is maintaining the minimum 34V on battery but you can see also much lower voltage at lower levels.... I remember something like 32V at ~30% battery level so maintaining the 34V at 50% is the overkill probably.
With the stock firmware is the decreasing peak power probably less noticeable.