BotoX/xiaomi-m365-firmware-patcher #31: Can't speed up more than 31.7kmph on custom firmware

tracer2535, 2019-01-07:
Hi, I'm trying to remove speed limit, I did set max speed to 45kmph and ticked "remove hard speed limit of 35kmph" - still I can never exceed 31kmph - I can feel that scooter manually slows down, so it's not the issue of not enough juice in the motor. When I go downhill - it slows down on 31. When I go in straigh line - it accelerates up to 31.5, then decelerates to 28, then accelerates again and so on. Any way to solve this issue?. Thanks

tracer2535, 2019-01-08:
Oh thanks!

clemensru, 2019-04-06:
I have experienced the same issue and I'm 100% sure that the limit is firmware related.

The power (voltage and amps) is definitely not the issue since the "slow down" also occurs when you let the wheel spin freely.

Having the wheel spin freely, it will first accelerate and exceed the speed limit for a short period of time, and then slow down to about 31km/h.

I have decompiled the firmware and was able to find the places which use the set speed limit for eco and normal mode and I've noticed some other potential limits - but I wasn't able to make sense out of it all yet.
I'm not that experienced with disassembling and the references to memory locations are a mess (in most cases pointing somewhere + some offset).
I'm not sure how the increased voltage elevates this limit, though - maybe the voltage is checked and the RPM is limited based on the current voltage.
Is there any additional information on the firmware, maybe a disassembled/commented version of it - or infos on findings? Which disassembler do others use? (I've used Ghidra which doesn't seem to support Cortex M3 binaries - so I had to modify the bin (ie the vector table) for it to make sense out of it)
I don't have the hardware to flash the chip if flashing via the app fails.. so I'm not too keen on flashing potentially broken firmware. :)
(I hope this comment gets noticed, or should I open a new issue for it?)

BotoX, 2019-04-06:
electric motor max rpm is limited by the system voltage
by using a higher voltage battery you can drive faster without any additional firmware patches (I get 39km/h on a full 12S battery)
you can stare at the firmware as long as you want, it's not going to defeat the laws of physics.

filipef101, 2019-06-11:
@BotoX Question, Is that rpm limit suposed to happen even on downhill?

BotoX, 2019-06-11:
yes, unless you turn off the mosfets completely like on 1.4 I guess
however driving too fast and then enabling them again will do hard regen and could destroy the electronics from over voltage
actually it could even destroy the mosfets from over voltage by simply spinning the motor very fast so it generates above 100V back emf lol