BotoX/xiaomi-m365-firmware-patcher #32: Cannot exceed 25km/h, even downhill

happyfrog99, 2019-01-31:
It doesn't matter whether I pick the Russian throttle algorithm, the regular one or 1.3.0, 1.3.4, 1.3.8 or 1.4.0 - I just can't get above 25 km/h.
I've set the power constant down to 40000, and while I did notice it accelerated faster, it still caps out at 25km/h.
When I'm going downhill I can feel that the KERS is kicking in to keep me at 25km/h.
At this point I've just run out of ideas - the one thing that I suspect might be causing problems is that I had updated my scooter to either 1.4.1 or 1.4.2 and then downgraded so I could use custom firmware - but I don't know how I would fix this.
I'm using GPS to monitor the speed

Nojevah, 2019-02-06:
Sorry for this privacy intrusion but what's your weight ? What about Temperature ?
You'll also need to take a screenshot of m365 tools, battery screen at 100%.
Considering weight/temperature/battery degradation, 25 km/h doesn't seem unusual.

The only strange thing is downhill. With 1.3.8 (and not 1.4.*), KERS disabled (at 42 km/h for example), NOT using throttle, you should be in free wheels and not feel KERS limitation. But if you use throttle again, yes engine braking will prevent you to overstep Vmax.

FWIW, with 65kg I'm going at 28-29km/h maximum in m365 tools but I would not be surprised I'm at 26 with a GPS. Try to check your top speed with m365 tools.

happyfrog99, 2019-02-07:
I'm 70kg so not too much more
I am holding the throttle down when going downhill so that's probably a part of the problem.
I'll see if I can grab the speed with m365 tools

L3o-pold, 2019-02-07:
I have the same kind of limitation but around 29-30 km/h (m365 tool). The M365 slow down around 27 km/h during 3 seconds and go back to 29-30 km/h and so on...
I am on a flat road by the way and it’s a brand new scooter.

Nojevah, 2019-02-07:
That's normal. There's some kind of limit (power/speed). When you reach it, power is reduced for 1-2 seconds. That's why people pick a slower Vmax. You're a little slower, but you have constant speed (better feeling/comfort). Depending on weight and battery state, you can choose 31/30/29 km/h.

chiastic-security, 2019-04-12:
That's a little disappointing... So there's realistically no way of getting faster than ~30km/h, even downhill?

How could the limit when you're going downhill be a battery saving thing?

Nojevah, 2019-04-12:
Yes you can. If you create a Botox firmware with KERS at 40 km/h (firmware 1.3.8 recommended) , you will be in free wheel downhill, you'll be able to go at 40km/h for example.
Careful though when you'll break at 40km/h, KERS will try to recover energy for batteries, and after 40km/h, I'm not sure hardware is able to handle it correctly (but maybe I'm too much shy/cautious).

chiastic-security, 2019-04-12:
Ok, thanks!

Is it possible to patch the firmware so that you can also exceed 30km/h with the throttle on? It's somewhat counterintuitive that you can go faster without the engine than you can with it.

Nojevah, 2019-04-13:
Flat surface: Even Usain Bolt can't go faster with his leg on a M365 ! xD
But I guess you were speaking of downhill: with throttle On, Engine break will prevent you from accelerating. No throttle if you want to go faster. It's the same as a car: neutral = free wheel. No patch is possible to change this behavior.

chiastic-security, 2019-04-13:
OK, I don't think this is engine braking in that sense, though. It's actively slowing down (I can feel it kicking in, and then turning off). Natural engine braking downhill would be much smoother, and just limit your terminal velocity.

The thing that limits the power for 1-2 seconds is presumably in the firmware somewhere, and could in principle be patched, even if no one's located that line of code yet?

gioo137, 2019-06-13:
Hi Guys, I have the same problem as "chiastic- security" user.
The annoying problem is in the plain. When it is loaded and up to about 70%
The speed is not constant. Over 30 k / H does not go and brake or power off motor for 1-2 second!!
In The botox site, there is "Remove hard speed limit of 35km/h"! I patch this but it doesen't work.
There is another speed control somewhere or the patch in botox site doesen't work fine.
Please help us. We understand that the scooter has the power but is castrated!!!!

Regards

Giorgi - ITALY
Xiaomi m365 CFW firmware 1.3.8 version ;)

davejupp, 2019-07-03:
With an ES2 front wheel and identical firmware I get the slow down behavior at 40km/h instead of 30/kmh with the standard M365 wheel. But it's otherwise identical, if you hit 40km/h (as reported by the pro dashboard, not sure of the real speed) you can feel it actively slowing you down until about 37km/h

This happens on both the pro and standard firmware.

Unfortunately running the ES2 front wheel tends to vibrate your battery apart cause it's lacking the inflatable tire.

Adorfer, 2019-07-03:
Is there any logic/controller in the front wheel which could make the difference?
Or is it simply the resistance of the coils(?) which lets the ESC going into throttle mode at a higher speed?
So in one way or another, this behaviour "downhill around 32km/h" seems somehere in the hardware and not fixable by the normale software (for the moment)

> Unfortunately running the ES2 front wheel tends to vibrate your battery apart cause it's lacking the inflatable tire.

I really don't like to derail, but no chance to mount an inflatable tire on the ES2 front wheel?

chiastic-security, 2019-07-03:
What is the reason for thinking it's not something changeable by the firmware, though?

Are we sure it's not just that no one's found the right bit of code yet?

Btw what happens if you change the parameter controlling the wheel size? Does that affect the speed at which it kicks in? (In other words, is it defined in terms of speed, or in terms of RPM?)

davejupp, 2019-07-03:
Unless I'm completely misunderstanding this, the speed is ultimately limited by the RPM of the motor which is fixed (for a set voltage). 
I suspect the M365 wheel is 500 RPM as 
wheel circumference = (2*pi*0.11) roughly =  0.69, 
and at 500 RPM that's 0.69*500*60 = 20.73km/h

Adorfer, 2019-07-03:
If the throttle-limit is hard linked to RPM: 

the circumfence of the ES2 motor/tire is 25% bigger than the standard M365 tire? 
(that could explaint 40 instead of 32km/h limit)

davejupp, 2019-07-03:
No, it's 0.5in smaller hub diameter, and the solid tire is smaller also. It just goes a lot faster (acceleration feels a lot quicker also)

Adorfer, 2019-07-03:
assuming that _"with all limit removed in current BotoX-tool"_ the M365 runs 40km/h on ES2-frontwheel motor: 

so the hard limit we are hitting on M365 (standardtire at about 32km/h) is not an RPM-limit in the ESC, but derives from the the sensing on/in the motor.  (coil resistance, number of poles, what else?)

BotoX, 2019-07-03:
google back emf
that's what limits your max speed
the motor acts as a generator
higher rpm = higher voltage
at some point the voltage difference between battery and motor is too small to produce any meaningful current so you can't go faster

chiastic-security, 2019-07-03:
No, it's not that. That would just mean it would asymptotically approach
top speed, and not go any faster. But it doesn't: it hits top speed, then
actively slows down for a few seconds, then the power kicks in again.

Something is actively managing the speed.

BotoX, 2019-07-03:
that's the software side of things freaking out when you try to push it beyond the limit
it can't sense any meaningful current and the control algo freaks out until there is current again

chiastic-security, 2019-07-03:
Is that in the firmware?

What happens if you nuke that bit of the code? Or more likely, change the
value at which it freaks out?

BotoX, 2019-07-03:
I haven't actually looked into that but I've seen some feedback stuff with current so that is likely to be that.
The issue can be simply avoided by setting the max speed to the correct max speed, eg. on 10S max of 31km/h. You can't go faster than that anyways.
Basically the speed control thing requests a current = delta * factor by doing delta = speed_want - speed_now
So higher difference in speed yields higher current.
And then another algorithm adjusts the duty cycle to try and reach that current
I guess if it tries to set 100% duty cycle and suddenly it gets reverse current values by the motor actually charging the battery or so it resets.

That's at least how I'd explain this behavior, no clue if it's that and how to fix it.

chiastic-security, 2019-07-03:
Something that makes me slightly dubious is that if this were the case then
I'd expect it to allow faster speeds downhill. But it doesn't: it maxes out
at the same speed downhill as on the flat.

Uphill, for me at least, it never reaches cutout speed. Again, this would
be a surprise if that were the issue, because I'd think it would cut out
but at a lower speed.

megabyting, 2021-01-28:
For increased the tyre's circumference, the maximum speed increased.  Since the final drive ratio is 1:1, this could be related to the rpm control.  The software could have a protection scheme to ensure the hardware (mainly for the MOSFETs) runs within the safety limit.