BotoX/xiaomi-m365-firmware-patcher #33: [NEW PATCH] 10" tires speed fix

truesurge, 2019-02-08:
Since 10 inch tires have larger diameter, angular to linear speed ratio is different.
M365 @1.4.0
ROM: **0x08006D44 MOVW R1,#345** 
And then it does tho MUL using R1. 
ES2 does the same, but using constant **390**
For 10 inch tyres value 315 work just fine,
Signature @1.4.0 **40 F2 59 11**
Fix with 315 const **40 F2 3B 11**

BotoX, 2019-02-10:
Nice, I'll look into adding this patch later.

noctrex, 2019-02-12:
Thanks a lot @truesurge and @BotoX ! This will be a great help for us 10inchers.On a side note, I've

truesurge, 2019-02-12:
There is, but it is not that easy.

BotoX, 2019-02-18:
Added the patch in 7a3cbe030cf0f72dbc2d073de7b80bc0e4723c85
Give it a try here: https://m365beta.botox.bz/

noctrex, 2019-02-20:
OK Thanks, starting trials today to see how it works.

noctrex, 2019-02-20:
Tried it out today, flashed the following parameters, with and without the wheel speed multiplier patch: