BotoX/xiaomi-m365-firmware-patcher #35: Hack Bluetooth

VVW44, 2019-02-13:
Hello,
Do you have an issue or a workaround to avoid the hack of the bluetooth connection ?
https://www.wired.com/story/xiaomi-scooter-hack/
Regards

davision, 2019-02-14:
That would be a nice feature for this custom firmware indeed.

BotoX, 2019-02-15:
Nope, that's an issue in the BLE module.
You can cut the Bluetooth antenna so it only works for like 10-15cm as a "fix" :p

robotsflo, 2019-02-16:
Hello, 
Can you post pic with mark of cutting zone ? 

Thanks a lot for your work to 365 users.

deg0nz, 2019-02-24:
I think another workaround is, that you have the scooter connected to an app every time you ride it. Just connect the scooter to the [m365 Tools](https://play.google.com/store/apps/details?id=app.peretti.m365tools&hl=en_US) app (Android) or the [M365 Dashboard](https://itunes.apple.com/gb/app/m365-dashboard/id1444998530?mt=8) app (iOS) before riding.

When I have an app connected, I'm _not_ able to connect another Bluetooth device to it, because the scooter is not visible anymore.

It would be nice if someone could confirm this with their scooter.

deeluuxe, 2019-03-15:
Is it possible to disable the BLE module? I would like to upload the customized firmware by ST-Link only to secure the scooter a bit.

deeluuxe, 2019-03-15:
...just to clearfy: With "BLE module" I mean the bluetooth connection only - I want to keep the dashboard circuit board of cause.

Dynamic0a, 2019-03-22:
I also want to know how you can disable the bluetooth..

VMBindraban, 2019-03-31:
Wrap the BTL module in tin foil?
If you ever need to connect just remove the foil.

BotoX, 2019-03-31:
[image]
[image]

See the antenna on the BLE board?
Just cut the trace somewhere to make it a short trace only, that'll cripple the performance of the antenna and reduce the range to a few centimeters.