BotoX/xiaomi-m365-firmware-patcher #38: Controller shutting down above certain amps

Hermit12, 2019-02-23:
Hi,
I use the m365 with 48V and custom made battery. I noticed that the scooter is shutting down above approximately 40kmh with a strong brake. At first I thought the bms is shutting down but that wasnt the case (looked into debug). After two blown controllers I put more solder on the down site of the controller so it wont blow now but it is shutting down. I can definitely say that this happens above 15A and below 20A. But during a ride the controller can handle 30A without a problem, only braking lets him shut down.
Is there a max amp limit in the firmware and if so is it possible to set it higher? Have you noticed anything like that?
If that doesnt work is it possible to dissable KERS completely?
Thank you

dalgibbard, 2019-02-23:
In a typical active breaking scenario with brushless motors, a back voltage can cause voltage spikes- it might be that you're exceeding supported voltage? Do you have any means to monitor voltage while experiencing the issue perhaps? I'm not sure what the voltage limitations of the boards are, but it's food for thought at least.

Hermit12, 2019-02-23:
Thats a good idea. I cant monitor the voltage during middrive/brake but I can set it higher than 54,6V (for 13s) and look if the problem still persists

BotoX, 2019-02-23:
13S is risky, especially on a full battery.
The maximum voltage the components on the board survive is 60 Volts.
When you hit the brakes the system voltage rises by around 5 Volt, 13 * 4.1V + 5V = 58.3 Volt
Back EMF spikes can be considerably higher since there is only one big cap (high ESR) on the board so some chips are definitely getting hit by voltages they are not happy at.

I'd recommend you to set your max speed to 35km/h and see if the issue persists.
If it does then 13S seems to be too much for the scooter, maybe only charge to 4.0V and try again.

Hermit12, 2019-02-23:
Cant set the charging limit to 4.0V, it doesnt work. I think my chinese charger is not ideal. Anyway I set the limit to 60V and it is a bit better, so I guess you guys are right. Would a 100V 1000uF cap do the trick? I like the high speed, thats mainly why I upgraded it to 13s...

Hermit12, 2019-02-23:
Forgot to mention that I use the ES2 motor. Its stronger and I think the Back EMF should be even higher. But I can really recommend it, total different riding experience.

Hermit12, 2019-02-28:
Replaced the cap with a 100V 1000uF version, put the voltage limit to 63V, problem still persists. Is it possible to enter even higher numbers for voltage? I know you chose 63V because its the highes possible voltage the standard cap can handle, but I like to go a bit further.

BotoX, 2019-03-01:
If the battery goes above that voltage limit the scooter just beeps, it won't shut down.
You should join the telegram group to discuss your issue: https://t.me/XiaomiM365Hacking

Hermit12, 2019-03-28:
hopefully figured it out. If you use different motor with stronger/larger neodym magnets, the build in mosfets are not suited. During brake one of the mosfets near the temp sensor gets really hot for a split second, this resulted in shutting down. Sometimes it even blew. I changed all of them with IRFB 3077 ones, they run cooler and are of better quality. So far no problems, if anything happens I will post it, if anybody is interested. Thanks