BotoX/xiaomi-m365-firmware-patcher #40: Can i prevent Lock to Sleepmode? (Lock disables after 3H)

softmarshmallow, 2019-03-11:
https://www.reddit.com/r/ElectricScooters/comments/azqmpm/is_it_possible_to_disable_m365_es2_sleep_mode/

https://en.miui.com/thread-5958454-1-1.html

Same problem with this. Can you Add feature to block Sleeping?
 To Quote
>If I didn't ride scooter in couple of hours, then M365 / ES2 powers off automatically even it's locked(M365: 3 hours, ES2:1 hour). After that, If I power it on again, then it's unlocked. I want my scooter keep lock mode and prevent to sleep even after 3 hours. Is there any way? Please share me some link, products or custom firmware which can handle this problem. Thanks.

openffchrome, 2019-05-06:
i see it have been added, becarefull with this setting if you use custom apps to lock your m365 because sometimes it bug and you have to wait 3hours for the unlock :s

softmarshmallow, 2019-05-07:
@openffchrome Ok so, did you find any solution to prevent this event?
Im working on creating external ble device to prevent this..

openffchrome, 2019-05-16:
no, i don't lock it anymore to avoid it...
what kind of external device?

neosarchizo, 2019-06-12:
@openffchrome I just want to disable the sleep function. But I can't find an option about that on your application. I think you know an address of the sleep function in ASM binary.

openffchrome, 2019-07-09:
@neosarchizo  i'm not the developper i don't know