BotoX/xiaomi-m365-firmware-patcher #41: Can't flash other firmware to m365

Andreix9, 2019-03-18:
Hi there.
Recently that I own a m365 xiaomi and I found a video where it explained how to flash a firmware which could be customizable regarding speed. I had a version 142 of firware so I had to downgrade to 140 and then flash the 138.
Now I wanted to slightly modify some numbers but the M 365g app says the firmware is not acceptable. invalid bin. I tried to get back to 140 bin but it doesn't allow me.I tried the binari from here 1.38 firware but I don't know the specifications on that firmware..
Could someone tell me please what should I do to be able to flash once again a customizable one ? please

BotoX, 2019-03-18:
generate firmware on https://m365.botox.bz, you get a .zip file, don't extract it.
use the latest m365 downg app from the play store, use the .zip file.

Andreix9, 2019-03-22:
Thanks. That worked. But it appeared like it was using too much battery, and with the update that poped out I thought this will solve it. Now that I tried the way home with the update and normal settings, it appears it was only because it’s just a long distance that climbs. 
Now I’m stucked with the latest firmware that I can’t downgrade from.1.5.1
Do you know what bin or zip should I flash to downgrade from latest firmware please ?

BotoX, 2019-03-23:
Downgrade is only possible with ST-Link right now.