BotoX/xiaomi-m365-firmware-patcher #42: Can't flash after update to 1.5.1

otacon64, 2019-03-19:
Unfortunately I've updated my scooter to firmware v1.5.1 and now I can't flash custom firmware. Is there any plans for support of v1.5.1?

raypettersen, 2019-03-31:
I got mine shipped with 1.5.1
Is anyone working on the problem?

BotoX, 2019-03-31:
Need to flash the BLE module with ST-Link -> https://t.me/XiaomiM365Hacking/28567

Nojevah, 2019-04-12:
The other solution is to wait. There are good chances new signatures/keys used by BLE 0.8.1 will be decrypted. If BotoX is still in business at that time, you can expect a new version. :stuck_out_tongue_winking_eye:

snackman1, 2019-04-28:
Just found the solution on how to downgrade the BLE back to 072.  Install the Beta version of M365DownG and it will allow you to flash the BLE072.bin.

https://rollerplausch.fandom.com/wiki/Flashing_Firmware

Nojevah, 2019-04-28:
And if you're patient enough (recommended), you wait for official release on Play Store.

snackman1, 2019-04-28:
Just checked Google Play and as of April 26 version 13 (the beta ver on rollerplausch.com was ver 12) is now available for download :)

guyromb, 2019-04-29:
You are awesome guys! 
Works perfectly fine now :-)

fabiohbarbosa, 2019-05-23:
Guys my scooter is the model 16133 European Black (2 Fuses).

Can I apply the BLE downgrade following the link?https://rollerplausch.fandom.com/wiki/Flashing_Firmware

Which model is your @guyromb ?

Thanks :D

MarlonMrN, 2019-05-29:
Yes you can @fabioh, almost nobody has the Chinese old 1 fusion version. And those who have it, cannot flash cfw.
You are fine, go for it.

zalgocykel, 2019-06-23:
> 
> 
> Guys my scooter is the model 16133 European Black (2 Fuses).
> 
> Can I apply the BLE downgrade following the link?https://rollerplausch.fandom.com/wiki/Flashing_Firmware
> 
> Which model is your @guyromb ?
> 
> Thanks :D

What happened to the link? It's broken :/

Anyone saved the content that was on that page?

fedya, 2019-06-25:
I need help here.
Segway app updated my bluetooth mode to something => 081
and now i need to push scooter button to pair devices.
Any idea how i can disable it? Want to downgrade back.

zalgocykel, 2019-06-28:
> 
> 
> I need help here.
> Segway app updated my bluetooth mode to something => 081
> and now i need to push scooter button to pair devices.
> Any idea how i can disable it? Want to downgrade back.

I finally found a working link here: https://rollerplausch.fandom.com/de/wiki/Firmware_flashen
Here is the bluetooth bin: http://rollerplausch.com/dl/BLE072.bin
Generate all kinds of FW (including stock): https://m365.botox.bz/

I had my firmware at 1.5.1, what I did (with M365downG):
1. Downgrade bluetooth version
2. Then I went from 1.5.1 ->1.4.3  (stock FW) -> 1.4.0 (stock FW) -> 1.3.8 CFW

wjurban002, 2019-10-22:
I have found that this thread is the only topic related to my current issue but does not solve it, I have a ninebot s. I literally won it off the snap on truck and installed the segway app and unfortunately updated the firmware BEFORE i knew of its limitations and AUTO BRAKING when you go over 9.9mph... thus i just figured out i have to hack this thing and whoops the bluetooth isnt working with flash apps.

This said. I have 1.6.5 and i understand I cannot flash it with this version. ES_downg is the only app that can see and connect to the scooter, but i am stuck. when i click detect version and no response occurs which prevents me from updating the ble. please help someone i just want this thing to quit auto braking.