BotoX/xiaomi-m365-firmware-patcher #43: updating and modifying the creation program

TugaM365, 2019-03-20:
Hello all the team there is the possibility to put the coded source of protection in the https://m365.botox.bz with the versions of the previous updates could have created us update Bin all with the protection bluetooth have can even add two batteries

They just need to extract the source code of the bluetooth protection of the 1.5.1 update to add in the creation program a location to be able to select or not the bluetooth protection with the firmware 1.3.8-1.4.0-1.4.1-1.4.2-1.4.2

Hermit12, 2019-04-02:
Other question, is it possible to flash the modified bin file with st-link?  Got a cheap replacement dashboard that doesnt work with camialfas app and need to flash for different baud rate.
Thanks

BotoX, 2019-04-02:
You should be able to write the modified firmware to 08001000 after unlocking the write protection with m365rec
Else take a look at https://github.com/etransport/py9b

Hermit12, 2019-04-02:
Thank you I will give it a try