BotoX/xiaomi-m365-firmware-patcher #44: Speed issues

KRomoda, 2019-03-24:
Hi all

First of all amazing to have this tool to make custom software. Great job.

I have installed second battery to my m365 and is all working well. Problem is that when scooter is fully charger it rides to 31kmph and then brakes and speeds up again. If battery is lower it is doing the same thing just limits to 28kmph. I have tried all lossible options and speed changes in software but scooter wont go faster. I can feel that there is enough power but it comes to one point and brakes again. 
Any solution for this?

Adorfer, 2019-03-24:
exactly the same here. 
perhaps it's not a brake when hitting 31kmph, but at least it feels like "full power cut". 
for my feeling it's like a regulator overshooting and then cutting back. 
 (and yes, i lifted the upper limit so something like 37kmph)

KRomoda, 2019-03-24:
Exactly. Tried few different speeds and motor strenghts but no change. The thing is that i did not had this issue like a month ago only with recent software.

BotoX, 2019-03-24:
set the max speed to 30 if you have a 10S battery (the default)

KRomoda, 2019-03-24:
I have stock battery and 10s4p in parallel. I do want  to go faster then 30 if possible

BotoX, 2019-03-25:
It's not possible with 10S.

KRomoda, 2019-03-25:
Ah ok. Thank you

KRomoda, 2019-03-25:
One more thing. I have now 2 charging connection, 1 on each battery. Is it ok if i plug 2 chargers in the same time? Is that safe?

BotoX, 2019-03-25:
Yeah that should be fine, just make sure the total charging current does not exceed 0.5 * total capacity.
That should be true for the small/cheap chargers.

KRomoda, 2019-03-25:
Not sure i understand. I have original and third party charger. Both 2A.

BotoX, 2019-03-25:
2A + 2A = 4A
original battery 7.8Ah + external battery 10Ah = 17.8Ah
17.8Ah * 0.5 = 8.9A

4A < 8.9A -> OK

KRomoda, 2019-03-26:
You are a star. Thanks a lot

closed 2019-03-26 by KRomoda