BotoX/xiaomi-m365-firmware-patcher #46: compatibity with xiaomi m365 pro?

pedroorellana, 2019-04-07:
Is this firmware compatible with the new ( 2019) xiaomi m365 pro ??.

The pro version has a LED screen with speed in the scooter, and also a more big battery. I don't know if use the same firmware as this.

frogale, 2019-04-07:
I have same question...

BotoX, 2019-04-11:
Maybe in the future.
They changed the compiler on 1.5.x so all signatures changed.
Would need to redo all the signatures, that takes a while and I don't even have a Pro scooter

lorek123, 2019-04-11:
I recently found this: https://www.spzshop.com/blogs/best-practices-and-tutorials-for-m365/m365-pro-already-hacked-no-speed-limit, maybe it is a good idea to contact this guy to see what he has.

Roki100, 2019-04-15:
> Maybe in the future.
> They changed the compiler on 1.5.x so all signatures changed.
> Would need to redo all the signatures, that takes a while and I don't even have a Pro scooter

im buying it tomorrow so i hope it will be soon! :D

chripede, 2019-04-29:
It looks like there's already a Pro version released by "the russians" http://xn--80atlli8e.xn--p1ai/ru/konstruktor-proshivok-pro

melgu, 2019-04-30:
Has anyone tried one of the two options provided here?

Ascarib, 2019-05-12:
Yeah, it works but m365hack's better.

Still turning KERS off doesn't work properly - it turns off it on brake! 

On 10" tires changing multiplier works only for BMS - the display is counting speed by its own for 8" :(

melgu, 2019-05-12:
My scooter still takes two or so months to get here. I will comment then.
I'm just a little bit scared to brick my scooter with faulty firmware.

tricade1, 2019-06-05:
I installed yesterday the GONKA on my pro without any problems. I can confirm its working (1.5.5). As @Ascarib already said the M365 hack is much better. When I use the BRAKE on the Pro you just use the rear wheel to stop the scooter there is NO help from the front wheel like its is with the Basic M365 hack.
Good luck @melgu !!!

By the way .... if you guys find a better way please let me know ! greetings from Germany,

tricade1, 2019-06-26:
Update - first Beta from Rollerplausch
http://rollerplausch.com/dl/DRV155-MitKERS_RUS_2A-50A.zip with KERS
http://rollerplausch.com/dl/DRV155-NEU_AUSROLLEN_RUS_2A-50A.zip without KERS

https://rollerplausch.fandom.com/de/wiki/M365_Pro_Dashboard_installation#Warum_keine_originale_Firmware.3F

I was able to install the 2nd one without any problems. Need to test the Scooter the next days.

klujanrosas, 2019-06-29:
hey @tricade1 after flashing that beta, it seems that the scooter does not longer go into charging mode when connected(it still charges though, but the LCD doesn't show as charging), would you know if this is expected?

tricade1, 2019-06-29:
Hi, I’ve not tested it yet.... let me have a look later !

th0mk, 2019-07-14:
I found this patch site for the pro: https://m365pro.scooterhacking.org/

tricade1, 2019-07-14:
Thanks that’s the same like the Russian website that chripede mentioned.

MelvinIdema, 2019-10-23:
Are there any updates 3 months later?

tricade1, 2019-10-23:
Not that I know ... looks like there is no update

roadrgo, 2020-03-22:
Anyone from UK or Europe that knows how to repair Xiaomi m365? Kindly write to me.