BotoX/xiaomi-m365-firmware-patcher #47: Cruise Control - sound

Sharky1231, 2019-04-08:
Is it possible to implement an adjustment to adjust or turn off sound when cruise control gets enabled?
Because I want to have cruise control enabled quite fast (after 3 seconds) but I dont want to be beeping around cyclists all the time, they are looking around, wondering what is happening, etc.

BotoX, 2019-04-11:
Yeah that could be done I think. I'll see if I can take a look this week.

openffchrome, 2019-04-11:
an option to disable the sound when turning on or off the m365 would be awesome also, i hate when everyone is wondering what is beeping :s thanks

hoerbi1000, 2019-04-19:
i would also like a change for no sound :)

BotoX, 2019-04-23:
Here you go, tested it only in my room, seems to work: https://m365beta.botox.bz/
Let me know if it works for you.

closed 2019-04-23 by BotoX

hoerbi1000, 2019-04-23:
i will check it in the afternoon :)

hoerbi1000, 2019-04-23:
Works ;)

commit 0eaa2289 (2019-04-23): Add cruise control beep disable patch, fixes #47
commit 8f38760a (2019-04-23): Add cruise control beep disable patch, fixes #47