BotoX/xiaomi-m365-firmware-patcher #51: Connecting Dashboard screen from PRO to 365

promotion, 2019-04-26:
Hi, 

Thanks for your site and tools, really cool i could tweak my scooter :) 

My question - do you think you could tweak it so we could install Dashboard screen from PRO version to standard 365 version ? 

(You can buy those on ebay for 16-20 EUR - like https://www.ebay.co.uk/itm/Dashboard-Bluetooth-Circuit-Board-Screen-Cover-for-Xiaomi-M365-M365-Pro-Scooter/223492587536 )

and there's video showing you can hook it to 365 using PRO firmware https://www.youtube.com/watch?v=WPBUHUKcQ1M with english subtitles. 

It would be soo cool if this could be doable :) 

If you need - i can buy you the screen for developing purposes - just let me know .

Thanks

gavinmcnair, 2020-01-14:
I had already installed the pro screen on my m365. When i installed a custom firmware the scooter performed far better but my display is stuck on 0 kph so does not appear to be compatible with the display

roadrgo, 2020-03-22:
Anyone that knows how to repair Xiaomi m365? Kindly write to me if so.