BotoX/xiaomi-m365-firmware-patcher #55: Instant mode switch unchecked = active

openffchrome, 2019-05-16:
hello i have the Instant mode switch unchecked but it's active, do you know why? it's a bug?

my settings :  https://m365.botox.bz?version=DRV143&normal_max_speed=28&eco_max_speed=5&motor_start_speed=4.98&motor_power_constant=46000&boot_with_eco=on

MarlonMrN, 2019-06-03:
Do you know for sure how that feature should work?

openffchrome, 2019-06-10:
of course, you can try out the settings, instant switch patch is not checked but firmware act like it's checked

Nojevah, 2019-08-16:
This option switches from normal mode to eco mode when driving (or vice versa). Are you sure that's what happening ? If this option is disabled, you can change mode only when stopped.

I don't even remembered if I activated this setting last time I created my firmware (3 months ago) so I can't confirm or deny. I know it was working some months ago.