BotoX/xiaomi-m365-firmware-patcher #57: Others Scooter

juabelgar, 2019-05-24:
Hi all! I have an SmartGyro Xtreme City that have the same components that the Xiaomi M365. Can I flash this firmware on it? The app M365_downG recognize the Scooter but when I try to check version, don’t show me nothing. 
The motor have 350W, the battery 7800mHa. I think that the firmware is valid for it, but I’m afraid of break it.

BotoX, 2019-05-24:
That's a shitty clone/fake M365, this firmware won't work with it.
Considering check version doesn't even work then you won't be able to flash it anyways.
There's a slight chance the components are 1:1 clones and you could flash the original firmware with an ST-Link.

juabelgar, 2019-05-25:
Ok! Thanks! I think that the “Check Version” option don’t works because the firmware version is different.  I'm afraid to download a cooked firmware and try flashing it.  I'm going to open the scooter and see what controller it carries and I'll tell you.
Thanks a lot.

juabelgar, 2019-05-25:
in general, they all carry the same components, with a different management.

Migu5499, 2019-06-15:
I have a clone too. I get the firmware of these chinese clone boards un a zip file if anyone wants to help decode the firmware would help us:) here are the files: https://mega.nz/#!r4BjXQBD!8_FTwRN9ccllJfa0n2q9C_oBrdK3yUm0CcyE0WFKego

juancornet, 2019-06-19:
> That's a shitty clone/fake M365, this firmware won't work with it.
> Considering check version doesn't even work then you won't be able to flash it anyways.
> There's a slight chance the components are 1:1 clones and you could flash the original firmware with an ST-Link.

Hi BotoX, I have just read your post. I have the same issue with a Xiaomi clone scooter. Would you expain me how to try these patch with the ST Link

roadrgo, 2020-03-22:
Anyone from Europe/uk that knows how to repair Xiaomi m365? Kindly write to me.

nextor11, 2020-03-22:
> Anyone from Europe/uk that knows how to repair Xiaomi m365? Kindly write to me.

You spammed a lot of people by posting this on multiple issues. I don't think this is a good way to ask for help. There are Facebook groups for this kind of requests for help.

You didn't even described your problem, so how can one know if he knows how to help you?

I suggest posting on this Facebook group, but be sure to include a description of the problem and what you did or what happened that led to this problem: https://www.facebook.com/groups/xiaomi.m365.electric.scooter/

roadrgo, 2020-03-22:
> > Anyone from Europe/uk that knows how to repair Xiaomi m365? Kindly write to me.

> ... described your problem, so how can.... 

Hi i want new batteries, i want also know what will happen if i change the front wheel (250w motor) to a new chinese 300w front wheel motor.

roadrgo, 2020-03-22:
> > Anyone from Europe/uk that knows how to repair

> (...)You spammed a lot of people (...)
> (...)described your problem, so how can(...)

Forgive me just need help I apologize. 
I want to know what will happen if the (250w motor) gets replaced with a 300w motor bought 
from ebay. I want to know what problems i will face from keeping the firmware modified on 
31 km/h, using it for one year, keeping it like that, Is it bad? What will/could happen?

roadrgo, 2020-03-22:
> > Anyone from Europe/uk that knows how to repair

> (...)You spammed a lot of people (...)
> (...)described your problem, so how can(...)

Just need help I apologize for that. I don't have Facebook. 
I want to know what will happen if the (250w motor) gets replaced with a 300w motor bought 
from ebay. I want to know what problems i will face from keeping the firmware modified on 
31 km/h, using it for one year, keeping it like that, Is it bad? What will/could happen?

roadrgo, 2020-03-22:
> > Anyone from Europe/uk that knows how to repair Xiaomi m365? Kindly write to me.
> (...)You spammed a lot of people (...)
> (...)described your problem, so how can(...)

Just need help I'm sorry for that. I don't have Facebook. 
I want to know what will happen if the (250w motor) gets replaced with a 300w motor bought 
from ebay. I want to know what problems i will face from keeping the firmware modified on 
31 km/h, using it for one year, keeping it like that, Is it bad? What will/could happen?

roadrgo, 2020-03-22:
Anyone from Europe/uk that knows how to repair Xiaomi m365? Kindly write to me.
(...)You spammed a lot of people (...)
(...)described your problem, so how can(...)

Just need help sorry for that. Don't have Facebook.

I want to know what will happen if the (250w motor) gets replaced with a 300w motor bought
from ebay. I want to know what problems I will face by keeping the firmware modified on
31 km/h, using it for one year and keeping it like that, Is it bad? What will/could happen?

roadrgo, 2020-03-22:
> > Anyone from Europe/uk that knows how to repair Xiaomi m365? Kindly write to me.
> (...)You spammed a lot of people (...)
> (...)described your problem, so how can(...)

Just need help sorry for that. Don't have Facebook.
I want to know what will happen if the (250w motor) gets replaced with a 300w motor bought
from ebay. I want to know what problems I will face by keeping the firmware modified on
31 km/h, using it for one year and keeping it like that, Is it bad? What will/could happen?