BotoX/xiaomi-m365-firmware-patcher #58: #beta poweroffs at 25% battery or less

MarlonMrN, 2019-05-29:
I have flashed the beta version from m365beta.botox (rollerplausch), it looked amazing, so much smoother and better to drive.

However, when my battery hitted 25% or so, when lauching or going uphills, the scooter keept turning off. 

I'm not sure if it was already fixed, don't know where to look the changelog for the beta version 

Nice work though, I will try to do my best when I have a free time to analize the code and do something on it ;)

BotoX, 2019-05-29:
Your bms shuts off because of undervoltage.
Charge your battery lmao.

MarlonMrN, 2019-05-29:
It's on 25%. I want to be able to use it until 3% or less.
 Makes no sense to lose 1/4 of the battery.

BotoX, 2019-05-29:
The % stuff is not accurate at all, it's just the voltage as a % value. And in the lower end of the voltage the discharge curve is extremely steep, so if you pull a lot of amps the voltage drops down quickly and the BMS shuts off.
To avoid this switch to eco with lower power.
The original firmware reduces power drastically to combat this, in the beta I had to remove that for the custom amp values.
And you have at most maybe 5% at 25%.
Stop discharging your battery until it's sucked dry,  if you need more range add another one in parallel or build a bigger one

MarlonMrN, 2019-05-29:
It's possible to change the under voltage limit?
It wasn't weak or anything(it was pretty good to ride still) was just
turning off.

I believe it could run a little bit more haha

I was on like 14km or so, on the stable version I can do like 22km

What are the impacts on having a lower voltage limit to poweroff?

I believe mine drops to almost 32v with low battery and the stable version.
No shutdowns so far, my battery is in like 5% right now (weak as f), but
still Rollin

Marlon Griebeler da Hora

The % stuff is not accurate at all, it's just the voltage as a % value. And

MarlonMrN, 2019-06-03:
Sorry for the misleading.

I flashed beta again today and the problem isn't exactly the one I reported.

Isn't a complete shutdown.

My scooter right now is on 20% and 34,9v

When I force the motor (uphill or starting), the power to the motor cuts off. the scooter remains on.

After some seconds or sometimes faster, the motor get power again.

Looks like the motor energy cutoff its using a higher voltage than the actual power cutoff (which turns off the scooter)

Adorfer, 2019-06-03:
so the remedy would be: do not use a low power divider and stay at the stock value. Or ride on eco mode when the battery voltage is low.

BotoX, 2019-06-03:
Looks to me like it's working as intended.
It's stopping you from murdering your battery.

MarlonMrN, 2019-06-03:
I checked the graphs and it's indeed when the battery voltage is under 32v (even if it's 'stable' at 34 or 35 and drops to 32 only when I press the throttle to the end)

The question is: why is there a limitation in supplying power to the motor under that voltage?
If the battery is critically low, isn't the BMS responsible for the entire scooter shutdown?
I don't believe there should be such a limitation.

I figure maybe that the motor could be spinning faster (in volts) than the battery is supplying and the power cuts off. Something like what happens when you are downhill and the wheel spin faster than the battery voltage and the motor ""shut downs"" the power.

Adorfer, 2019-06-03:
now you describing a third scenario: _"shutdown to overvoltage during downhill"_
this can probably avoived by removel of the hard 32km/h limit plus removal of the charge detection ("y-cable hack")

Can you please re-assess/reproduce the EXACT scenario?

MarlonMrN, 2019-06-05:
I'm not saying that my scooter shutdowns downhill.

Just that the motor stop using force when reaches a high rotation (probably limited by voltage?).
For example: when I get faster than x speed, the power consumption goes really low and the motor isn't accelerating anymore. That's what I called "motor shutdown downhill".

Adorfer, 2019-06-06:
what kind of "poweroff" are you describing now? Is it still the initial one a different effect in your last posting? (for that i would link to several other issues, describing this effect. But i would not associate this with the title "poweroff")

MarlonMrN, 2019-06-07:
Adorfer, the "poweroff" is the motor that dies.
Dies = stop accelerating