BotoX/xiaomi-m365-firmware-patcher #60: Wheel spin by its own.

MarlonMrN, 2019-06-08:
I'm using m365 beta firmware.

When riding the scooter, sometimes at low speed (something that I can walk alongside the scooter), I jump off the scooter. To get over a kerb or so. I never completely stop the scooter. (The automatic"pilot" is on, but not active at the moment)

What happens is this:
The wheel starts speeding and spinning by its own. If I press the throttle very slightly, it spins SLOWER . If I let it go, it keeps accelerating 
It just stops after a brake or if I force the wheel to the ground.

How to reproduce the scenario: 
- turn on the scooter (lol)
- push it to your motor min speed and slightly accelerate (just to the motor turn on)
- let go of the throttle and wait it slow down until you can jump off the scooter safely
- as you jump, lift the front wheel. It should keep accelerating and spinning faster

BotoX, 2019-06-08:
Known behavior if you disable KERS.
Might fix this sooner or later, it's not a big issue though if you stop the scooter completely.
Or just lift the front wheel and hit the brake for a second if it does happen.

BenasPaulikas, 2023-06-08:
2023 and also having this issue. Stopping is super smooth tho :D