BotoX/xiaomi-m365-firmware-patcher #61: Max phase/battery current does not work while Russian throttle algorithm (from CFW) is set

jolie16, 2019-07-05:
Hello.
Parameters “Max phase/battery current” work fine without “Russian throttle algorithm (from CFW)”. But when I switch on “Russian throttle algorithm (from CFW)”, parameters “Max phase/battery current” don’t work. I will try to explain why.
“Russian throttle algorithm (from CFW)” change a function that count current while we press accelerator. 
Here is a screenshot with changed function, from 1.3.8.
[image]

Code that calculate current in CFW:
Normal mode
`active_current = 150 * accel_lever_pos`
Eco mode
`active_current = 100 * accel_lever_pos`
accel_lever_pos is current position of acceleration lever. It can be between 0 and 125.
This function doesn’t use max phase/battery current parameters. In this case, max current in normal mode is 150 * 125 = 18750 mA and 100 * 125 = 12500 mA in Eco.
And there is a speed limit in Eco mode. 
`if ( current_speed > 6900 ) active_current -= 10 * (current_speed - 6900);`
6900 is a speed 20kmh (6900/345=20). On every kmh above 20kmh it will be reduce current by 10*345=3450mA.

Also I saw code in original firmware that can be same as CFW algorithm. Maybe it will be useful.
[image]
On a screenshot I marked this code in red.
Before if-else a variable `word_200007D6` is set to 0. And then it checked for two conditions:
`if ( word_200007D6 )`
and
`if ( word_200007D6 == 1 )`
When `word_200007D6 ==0` it work in base mode.
When `word_200007D6 ==1`, a next code executes:
`result = phase_current_1 * accel_lever_pos / 125;`
This code almost same as in CFW algorithm, but it uses phase current parameter.
Do not pay attention to the `word_200003F0`, this is a decompiler error.

Screenshots are from 1.3.8 firmware, and infomation about addresses on scrrenshots:
- module base address `0x80000000`;
- acceleration functoin offset `0x00000A68`;
- `0x20000000` is a RAM section.

BotoX, 2019-07-05:
Yes that is very well known but I don't use or like the russian algo so I'm not planning on fixing it.
For me it's just much jerkier and doesn't have any smoothing like the original algo.
I only added it because people were requesting it. If one wishes to use it and change the power then they can use the non-beta site.
I like the xiaomi one much better and it works totally fine for me and the majority of other people so I don't see a reason for spending my time on fixing this.
Though I suppose I could offer a way to change those constants (150, 100 and eco speed) in the russian throttle patch.

jolie16, 2019-07-05:
> Though I suppose I could offer a way to change those constants (150, 100 and eco speed) in the russian throttle patch.

That would be great.

Adorfer, 2019-07-06:
What is the result consequence from "this not working as expected/intended/ntbf"?
(I do not understand the impact of this issue.)

lucidnx, 2019-08-13:
Hello! I would like to have russian throttle with different power but I also like breaking feature from beta. Do you think that this should be added to stable firmware generator? stock brake is too jerky

kociu18, 2020-04-06:
> 
> 
> Hello! I would like to have russian throttle with increased power but I also like breaking feature from beta. Do you think that this should be added to stable firmware generator? stock brake is too jerky

Same to me

it would be great if the “Russian throttle algorithm (from CFW)” worked in the beta generator or if the braking force and speed could be changed independently in the normal version of the generator

LGM33, 2020-04-13:
> Yes that is very well known but I don't use or like the russian algo so I'm not planning on fixing it.
> For me it's just much jerkier and doesn't have any smoothing like the original algo.
> I only [...]

> Same to me
>
> it would be great if the “Russian throttle algorithm (from CFW)” worked in the beta generator or if the braking force and speed could be changed independently in the normal version of [...]

Can this please be fixed? The non-beta site can't change regen braking params. I have disabled the physical brake and only use electrical brake. It is very smooth when minimum brake phase current is set to 1. M365 now brakes just like the Ninebot ES4.

I would like to use smooth brake at the same time as russian throttle. However, I can't change power constants on the beta site. Current control is way smoother at slow speeds and offroad, but has lower torque compared to speed throttle at the same power constant. This makes russian throttle pointlessly slow in the beta patched firmware. Just re-add the modify power constant option below russian throttle. No need to make fine tuning phase and battery currents compatible with russian throttle, just re-add a power constant override.

Thanks

LGM33, 2020-04-26:
The m365 pro firmware patcher has this issue resolved by using powerconstants and the Xiaomi current throttle algorithm.

`this is the patch for enabling the built-in alternative throttle on M365 Pro
`    def alt_throttle_alg(self):
        sig = [0xF0, 0xB5, 0x25, 0x4B, 0x25, 0x4A, 0x00, 0x21]
        ofs = FindPattern(self.data, sig) + 6
        pre, post = self.data[ofs:ofs + 1], bytearray((0x01, 0x21))
        self.data[ofs:ofs + 2] = post
        return [(ofs, pre, post)]`
i'm pretty sure it can be adapted with little to no work for the regular Xiaomi M365 `

lothean-dev, 2020-04-26:
> this is the patch for enabling the built-in alternative throttle on M365 Pro
> ` def alt_throttle_alg(self): sig = [0xF0, 0xB5, 0x25, 0x4B, 0x25, 0x4A, 0x00, 0x21] ofs = FindPattern(self.data, sig) + [...]
Posted that info on Discord, and after further research, I can say that it's located at ROM:08001A6C for DRV134.

LGM33, 2020-04-26:
This is a tested patch to enable current throttle while being compatible with increased power. (Power constant override, not phase current override.)
From user Lothean.

`    def alt_throttle_alg(self):
        sig = [0xF0, 0xB5, 0x25, 0x4A, 0x00, 0x24]
        ofs = FindPattern(self.data, sig) + 4
        pre, post = self.data[ofs:ofs + 1], bytearray((0x01, 0x24))
        self.data[ofs:ofs + 2] = post
        return [(ofs, pre, post)]`
this is the patch i have tested and that seems to work

LGM33, 2020-04-27:
> Same to me
>
>
>
> it would be great if the “Russian throttle algorithm (from CFW)” worked in the beta generator or if the braking force and speed could be changed independently in the normal version of [...]

Update: this issue is now fixed on ScooterHacking’s branch of firmware patcher.
Use it here: https://m365.scooterhacking.org