BotoX/xiaomi-m365-firmware-patcher #63: i got 1.5.2 firmware, will i be able to use downG ?

sl4p3z, 2019-07-15:
i got 1.5.2 firmware, will i be able to use downG ?

BMS: 126
BLE: 81
DRV: 152

i won't do shit :( i heard some people saying that 1.5.2 doesn't work anymore

BotoX, 2019-07-15:
try https://t.me/XiaomiM365Hacking/48462

sl4p3z, 2019-07-15:
> 
> 
> try https://t.me/XiaomiM365Hacking/48462

I was already looking for their support, was about to connect thanks and sorry for coming to the wrong place

GizmoTheGreen, 2019-07-22:
DownG worked for me, only issue is I am experiencing speed fluctiations in motor when riding over 25km/h?
eco mode is fine but normal I set to 30km and it's kinda jerky on max speed. I don't know if I must downgrade BMS or something too?

I tried both 1.3.8 and 1.4.3, I think 1.3.8 actually felt a bit worse...

Adorfer, 2019-07-27:
> DownG worked for me, only issue is I am experiencing speed fluctiations in motor when riding over 25km/h?

this is not an issue with 1.5.2 firmware or the downgrade-Process. 
Please have  look at the following issues, answering your question: 

https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/25
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/32
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/31
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/62
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/58
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/44
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/15
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/12
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/67

GizmoTheGreen, 2019-07-28:
> > DownG worked for me, only issue is I am experiencing speed fluctiations in motor when riding over 25km/h?
> 
> this is not an issue with 1.5.2 firmware or the downgrade-process.
> Please have look at the following issues, answering your question:

So basically I got my speed set too high? I think I did 31 cause it said to buy one over the actual speed I wanted. and people are saying 30 is the limit of stock motor without shortening its lifespan.
more testing between 1.3.8 and 1.4.3, the latter has the fluctuations much less noticable. will, try setting 29km/h and see if ti goes away.

Adorfer, 2019-07-28:
The effects you describe do not have any relation with 1.5.2 firmware or the downgrade process itself.