BotoX/xiaomi-m365-firmware-patcher #65: Question BMS 10s4p

Frederikom365, 2019-07-25:
Hi
I have a 10s4p 12.8Ah battery without BMS.
I want to make it in my m365 in place of original batterie.
I juste have to put a generic BMS?
I can flash the BMS to make it compatible and how?
Can i keep the original mother board and Dashboard ?

Thanks 👍

BotoX, 2019-07-26:
A generic BMS won't work, it has to be able to talk to the scooter.
I've made open source firmware for a chinese smart BMS that works with the scooter: https://github.com/BotoX/xiaomi-m365-compatible-bms

Frederikom365, 2019-07-26:
Thanks for the answer.
I Order the bms and Will try to install it with my battery 
[image]
Question , For a 10s battery i have only 8 wires on the white plug and 4 on the black plug , is it normal ?
Thanks

Frederikom365, 2019-07-28:
Ok thanks for the the answer.
My 10s4p (12ah) and the original 10s3p have the same voltage so i can  keep the original bms no?

BotoX, 2019-07-28:
That'd work but the original bms is pretty flimsy and good luck finding the balance connector.
Why don't you connect the batteries in parallel?