BotoX/xiaomi-m365-firmware-patcher #67: Momentary loss of power after reaching top speed, with extra batteries attached

nge6427, 2019-07-26:
I recently added two 10s4p 12Ah batteries in parallel, using the Y method (W, in this case). After doing so, there seems to be an issue where the power will momentarily drop once a certain speed is reached. This starts around 80% and gets progressively more noticeable and happens at lower and lower speeds. I tested with both regular and beta firmware, on 1.3.8.

At 80%, it'll reach 18.5mph or so, then I'll feel a brief brake, it slows down to 17.5, and picks back up. Basically sinusoidal. Around 60%, this happens closer to 18mph, and the loss of power is a bit longer. Not using the russian throttle algorithm btw.

My suspicion is that the speed isn't supposed to get that high normally at such a percentage, but the external batteries don't understand that and keep pumping power, until the stock BMS decides it's had enough? Maybe I should connect the batteries directly to the stock one, so that all power goes through the stock BMS instead of just 1/4th of it. The externals have their own BMS so this shouldn't cause issues if I'm correct.

My temporary solution is to just have the max speed on Eco set to a couple MPH lower, and switch to Eco once power drops below 80, but the issue will still come back eventually once the batteries get low enough.

Would appreciate any insight you might have.
Thanks.

BotoX, 2019-07-26:
On 10S the max speed can't be set higher than 30km/h else the problem you are describing appears.
The exact reason why it's kinda wonky is not known but the speed limit comes from the motor working as a generator.
Higher motor RPM = higher voltage generated -> lower voltage difference between motor and battery -> can't supply enough power anymore

nge6427, 2019-07-26:
So I understand that back EMF is ultimately the reason as to why the top speed decreases as the battery gets more and more drained, but what I don't understand is why this issue only appeared after adding the external batteries. 

When I was running on just the stock battery, at 100% charge, the scooter would reach my set max speed (30km/h at the time) and just stay there. As the charge fell, it would just reach progressively lower speeds but it didn't lose power once and oscillate once it hit those speeds. For example. at 50% it would reach about 27km/h and just stay there.

But now, even if I have my max speed set to 30km/h as you advise, once the charge gets low enough, it'll do that oscillating power loss thing before getting to 30, since it not actually supposed to hit 30 at that charge (at least with the stock battery).

Am I wrong to think that it's due to the power being delivered from the external batteries not being measured by the stock BMS and therefore not being limited? I don't know how I'd fix it otherwise, other than setting max speed to the top speed the scooter can do on say, 25% charge.

BotoX, 2019-07-26:
I don't think I've heard yet that someone is having issues at 30km/h, even less so with an additional battery. Could you share your firmware settings?

nge6427, 2019-07-26:
When I was doing these tests with the additional batteries, I had normal set to 32, and eco set to 30, with the currents set to 27000/54000 for both eco and normal (wanted same power output but different speeds). 1.3.8, KERS 40, motor start 0.5, remove 35 limit, remove charge mode.

I can use normal all the way down to 70% or so, at which point I have to switch to eco to prevent oscillation, but even then, that's only good down to about 55% at which point I start to get very faint but noticeable oscillation if the road is very flat and smooth.

When I only had the stock battery, I had normal set to 32 (sometimes 33 if I had a chance to charge fully beforehand) but didn't have this issue at all. Was using non-beta, power constant 40000-something (DYoC). I'd hit 32/33 at full charge, down to 28 around 50%, and down to 26 around 25%, but it would never oscillate (presumably because of lower power output at lower battery percent?)

If you want, I can make a few laps around the block and record the graph with m365 tools if that'd be of any help.

BotoX, 2019-07-26:
I guess you're going to have to live with that, very faint oscillation at 30km/h sounds normal.
That's probably the algo that controls power, it's power = (speed_want - speed_have) * acceleration
So it keeps on changing the power all the time slightly and you can feel that when it's flat and there is no wind

Adorfer, 2019-07-27:
i guess we need an faq, otherwise this topic will be reincarnating over and over again.

It's just different how people rephrase "loss of power", "shutdown", "oscillating top speed" or "current battery voltage limiting top speed even downhill"

the folloing are just a few, but basically ALL boil down to the same scenario and do have explanation. (perhaps i missed some issues (mis-)labeld as "A question" or "only 2 questions")

https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/25
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/32
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/31
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/62
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/58
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/44
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/15
https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/12

nge6427, 2019-07-27:
I don't think my question is a duplicate per se. I know *why* there's a top speed limit, the question was why did the oscillation start happening after I added batteries in parallel, and didn't happen on the stock battery, without touching the firmware (at first). That's why I asked if the BMS had anything to do with it, on account of only 1/4th of the total current going through the stock battery, so I was wondering if that was messing with some metrics.

Adorfer, 2019-07-27:
From my experience with external batteries wired in Y configuration: some feedback loops become more visible/less smooth. Some of the issues might be mitigated by running lower AWG, others are enforced. 

e.g. if you have due to lower total battery Ri a higher acceleration, even when reaching for vmax/vcuttoff: You can feel the cuttoff (the moment "loss of force") more drastic. 

or in other words: The effect is the same as before, just more visible. 

> didn't happen on the stock battery, without touching the firmware (at first).

How did you reach 31km/h without touching the firmware?

BTW:

nge6427, 2019-07-27:
Sorry, poor phrasing on my part. What I meant was that I didn't change the firmware after adding the batteries, and the oscillation appeared.

I have them wired with about 3 feet of 14 AWG each. I changed from a Y config to soldering directly to the stock battery pads, since I wanted to get accurate power readings (instead of just the stock's power output).

So my understanding is that if I lower the max motor power (instantaneous), the oscillation should be less noticable?