BotoX/xiaomi-m365-firmware-patcher #68: BatteryPack Upgrade

ltwin8, 2019-07-27:
Hello, I plan to change the cells of the internal battery pack to ones with 3500mAh 10A cont. 

Can I change the scooters original BMS firmware so that it knows the new capacity and goes down to 2.8V per cell?

it would be very nice if you can implement this in the configurator, or just upload a file like 
10500mA instead of 7800mAh
2.8V per Cell cutoff

thanks for your time!

Adorfer, 2019-07-27:
As layed out in https://github.com/BotoX/xiaomi-m365-firmware-patcher/issues/19
this firmware is about the ESC, not about the BMS.
If you want to modifiy the working of the BMS: This would be a different project. 
More likely you would then use some compatible opensourcefirmware-BMS, which is greatly reconfigurable. 
(The need of your modification to the BMS in a 10S-setup i do not see, but you will have your reasons. But off course out of scope, as stated above)

ltwin8, 2019-07-28:
Yes I read that already, however, is there any source for the code of the original BMS? I would like to keep it as stock as possible and since the voltage and current stays the same old BMS should work, only capacity is greater than.

roadrgo, 2020-03-22:
hi itwin, someone wrote that 3500mah cells are dangerous, not compatible, to put inside m365 Xiaomi. Is it true? do they get too hot? Which cells do you use? Do you know how to repair? which country you live in?

ltwin8, 2020-03-22:
I am from Germany.

I use INR18650-35E and yes, I know how to repair, solder, spotweld, Design circuits etc.

I have the scooter on most aggressiv settings (especially Brake) and no problem at all.

I put the 3.5Ah cells in right when I got the scooter, everything runs fine.
Only the BMS cuts off way too early (3.2V) but it seems no one has any idea and can’t publish CFW for BMS which is pretty sad.

I keep original BMS because it fits perfectly into the body.

roadrgo, 2020-03-22:
> [...] repair, solder, spotweld [...] 

I'm from Sweden, my greetings to Germany 🙂

Why did you install 3500mah cells on the first day?
The range for M365 (notpro) is 30 km. With the INR18650-35E cells, 
how many kilometres did you get range? Aggressive settings, do you 
mean modifying the firmware with DownG on an Android phone? 
I don't have Android, I have iPhone, how can I set mine on Aggressive?
What's your top speed now?

ltwin8, 2020-03-22:
Greetings to Sweden !

Firstly with android of a friend, meanwhile doing exist in iOS too.

For me range was factory on highest speed under 20km. Now I get well over 20km with this update.

I used the pro Firmware because I added the Pro dashboard (display) and set everything high, the breakcurrent to highest, KERS to 3Kmh, motor current to I think equivalent to 900W, and importantly: added thermal compound in the ESC.

My topped on 10S is 30kmh, but will modify 13S for around 40kmh

roadrgo, 2020-03-22:
>[...] exist in iOS too [...]  pro Firmware [...]  set everything high [...] motor current[...]  900W [...]  thermal compound [...]  ESC [...]  on 10S is 30kmh [...]  13S for around 40kmh 

What do you mean with iOS too? I downloaded DownG on my iPhone and it did not work, i looked everywhere people are writing everywhere that modifiying on iphone is not possible? Can you help me please? Pro firmware? You mean you never used DownG app? How did you do it? I think i did a mistake ordering an android phone yesterday 😧

ltwin8, 2020-03-22:
I used downg on android from a friend before, because it was not in the apple-appstore, now it is and I used my iPhone to download firmware to the m365.

CFW for scooter: http://сяокат.рф/ru/konstruktor-proshivok-pro
Pro Dashboard: around 10€ from AliExpress, flashed with STLink adapter to new firmware to be able to flash all (DRV BMS BLE) with bluetooth.

roadrgo, 2020-03-22:
my firmware version is 1.4.2, which cfw should i download if i want to hack it? to go as fast as possible? I pay you money, write your whatsapp, i paypal you for guiding me.

ltwin8, 2020-03-22:
What exactly is your setup until now?
Anything modded? Or all factory?

roadrgo, 2020-03-22:
> What exactly is your setup until now? Anything modded? Or all factory?
 Everything is factory and i have not changed a thing with it. I got it 4 days ago.
 It says on Xiaomi home: Firmware version V1.4.2.

roadrgo, 2020-03-22:
It's asking me on the app if i want to upgrade, should i?

roadrgo, 2020-03-22:
Here my whatsapp : +46763967844

ltwin8, 2020-03-22:
Hello, no, do not upgrade please. (Encryption.)

roadrgo, 2020-03-22:
Remove all the conversation here when we finish we are using this place like a messenger. 

"Base version of your firmware" (there is no less that 1.5.2 to chose which one should i choose?)

ltwin8, 2020-03-22:
Hello, I can not find your number in WhatsApp.

I use also iOS, so maybe we can use there messenger

roadrgo, 2020-03-22:
Made a mistake, the whatsapp reinstalled it was stuck with my older number, try now again text me on +46763967844 should work im sure

roadrgo, 2020-03-22:
Try now again on the same number on whatsapp +46763967844