BotoX/xiaomi-m365-firmware-patcher #70: Beta: "brake lever params" explanation

Nojevah, 2019-08-16:
Hello, I'm sorry if you had planned to add explanation later, but I'm quite tempted to test this beta (even if I don't know what could be improved with the actual firmware).

I don't understand at all what these settings mean, except it concerns engine breaking:
Lever virtual limit (1-130)
Minimum phase current
Maximum phase current

I could do like I did with other settings: try and compare. But I don't even know if there are dangerous value for my security or even the scooter's safety.

From my experience, I know engine brake kicks in 1 second after using break handle. I was surprised at first before to understand that it is for safety reason (not be thrown above scooter).
Anyway I don't really like engine breaks in general because i think it damages engine (and even batteries for scooters with KERS).

Do you advise some testing values to understand how it works ? Or values not to use ?

Addendum for the beta page:
- The default normal max speed is at 28. I think it's a mistake, it should be at 25.
- It lacks a warning not to abuse for max battery current (for those who don't understand what they are doing): the old 32000 should now be 27400 (or 27000). I don't know for max phase.

BotoX, 2019-08-17:
The max speed in normal is actually 28 in the firmware.
Lever virtual limit: you can make it do 100% current earlier
Like normal the max current is at about 2/3 of the lever pressed, you can make the range smaller with the limit
Min/max phase current should be self explanatory, you set the max regen phase current
Battery current should be about a third from that
Use lower start value for smoother beginning and higher max for more power

Nojevah, 2019-08-17:
Ty for this answer though I'm not sure I understood.

Lever virtual limit: The higher the value, the sooner "max phase current" will be reached when pressing brake handle ?

Min/max phase current: Admitting I don't want at all engine brake (for testing or measuring disc brake efficiency), I can put 0 to both min/max values ? I understand it is not recommended for safety reason, but is it dangerous for scooter engine/batteries ?

(Sorry for 28km/h, I was sure the default was 25 for months !)

LinoBarreca, 2019-09-06:
> Lever virtual limit: you can make it do 100% current earlier
> Like normal the max current is at about 2/3 of the lever pressed, you can make the range smaller with the limit
> Min/max phase current [...]

I didn't understand this answer either.
What is "Lever virtual limit"?
The position of the brake lever at which regenerative brake starts (with minimum current)?
Is 1 "completely released" (or close to it)?

gunnerysurgeon, 2020-01-18:
i believe the virtual limit means the position where max engine brake is applied.  so for example if you set it to 60 (range is 1-130), it would apply full engine brake at about 50% pull of the brake lever.