BotoX/xiaomi-m365-firmware-patcher #71: Beta: "Speed control params" not equal to default (non beta) "power constant"

Nojevah, 2019-08-22:
I've done comparison tests with another scooter (side by side, nearly same weight) and there was a difference, then I checked speed with m365 Tools app, and the difference is confirmed.

My settings:
default firmware: 1.3.8; vmax 30km/h; Motor power constant: 42000
beta firmware: 1.3.8; vmax 30km/h; max battery current: 21000; max phase current: 42000
So all in all, it should be the same or beta slightly quickest.

In general, in the same straight line:
With default firmware, I'm between 28 and 29 km/h
With beta firmware, I'm between 27,4 and 28,4 km/h and it looks like acceleration is not as powerful (but no evidence, so subject to bias)

I've tried another firmware with beta: vmax 31km/h; max battery current: 15000; max phase current: 30000.
With this one, I'm between 28,4 and 29,4 km/h. But I can't use it, since at the end of the ride my battery can't reach vmax so I have the classical jolt (despite using low acceleration settings).

Do you think it could be possible to allow intermediate Vmax like 28,5 or even a new bigger constant like: 3000 = 30km/h; 3200 = 32km/h etc. ?

BotoX, 2019-08-23:
There is an acceleration constant which can provide more power at speeds close to the max.
The beta current limits are used for this:
Vdelta = speed_want - speed_cur
Current = Vdelta * acceleration
If current > max_current:
    current = max_current

So when the two speeds are close the delta is small and the current isn't big enough.
So I guess a patch should be added to improve this
But making this acceleration value larger also makes the throttle less smooth I think, gotta try

Nojevah, 2019-08-24:
Thanks for this detailed answer.
In my case, "acceleration" is not what I'm searching. My goal is really to have the max possible value for "speed_want", considering at the end of the ride, my battery will be around 38,70 V.
So if instead of 30 or 31km/h, I could use something like 30,5 I think it would match default firmware (and maybe use slightly less battery since less acceleration).