BotoX/xiaomi-m365-firmware-patcher #73: xiaomi m365 system/software stuck

Greyzers, 2019-08-25:
Hi! Could someone have the same problem that the system stuck? The light are just ON and the power button do nothing. Removing the pattery cable is the only thing what help to restart the scooter. And then worked again.  Please can someone help that problem? Now is the fifth time when I am stopped on the road far away from home. Scooter Always stopped working when the first battery light goes off around the ~72% of battery.

Thanks !

BotoX, 2019-08-25:
Which version, which patches, etc.....

Greyzers, 2019-08-26:
Firmware 1.3.8, KERS min 31. Max Speed 31, eco mode speed 27, motor 40165, start speed 1.1, no cruise control beep, no russian algorytm, 1 stock battery, no speed limiter controller patched.

roadrgo, 2020-03-22:
Anyone from Europe/uk that knows how to repair Xiaomi m365? Kindly write to me.

Greyzers, 2020-04-05:
The battery voltage is OK. 42v and there are no error codes. There is no way to connect scooter via bluetooth. And the bms controller led is solid red. The battery lights are on and the power button do nothing. Pluging the charger wont help ether.:(

Greyzers, 2021-08-24:
I changed the firmware back to the original. Now it has been working normal for a long time.

igorneumann, 2021-08-25:
Fixed my problem yesterday... It was a couple of cables of the panel that was caught by the screw when I replaced the panel.

It just happens to work fine for days and break just after a flash.

So hardware provlem (removing the cable and putting it back correctly fixed the issie.