BotoX/xiaomi-m365-firmware-patcher #76: BackEmf speed limit

guygil, 2019-10-16:
hi.
 I understand that BackEmf naturally limit the motor speed , due to self-induction .( to 30~31 with 10s, and 35~36kmh with 12s battery..etc ..)
 But not understand how FW detect it , and why brake and speed régulation is applied .I've analysed the voltage phase sensing on STM32 : noway . actually working on phase current sensing way :when at top physical speed, STM sense null or negative current via U1 while riding at full throttle ..not sure ..
 so .. is it possible to disable or limiting this frustating regulation ? by option in a new fw builder ?
Is it a fw security to avoid overvoltage on battery via mosfets's bypass diode ? 
thanks .  Cheers from France for your job ..

guygil, 2019-10-16:
seen your reply  on another post. .
'that's the software side of things freaking out when you try to push it beyond the limit
it can't sense any meaningful current and the control algo freaks out until there is current again"
it's really seem to be the current sensing..
~ no crurent with high  speed >problem > brake. .

cptX, 2020-03-29:
Yes, but why people that increase the voltage manage to get more speed? Theoretically more voltage will lead to more current and then the limit you describe would be exceeded again...

guygil, 2020-03-29:
same phenomen for friends with 12s or 14s . generally at full battery .. and of course at higher speed.  but not for all . motor's depending.
But at higher speed/voltage , current not necessary higher , because efficiency is higher .. not sure , because I've not the motor data ( torque/RPM/W/efficiency ..) I search it ..

danilo-2, 2020-03-31:
several possible explanations; There are slight discrepancies in the hardware .. Their Speedometers are inaccurate... They are exaggerating..   My penis is bigger than yours syndrome ?