BotoX/xiaomi-m365-firmware-patcher #77: Possible to make scooter work without BLE Board?

duffann, 2019-10-21:
If it is possible, I would like to know how according to your experience ?

BotoX, 2019-10-21:
The BLE is the master on the bus, it sends the throttle and brake signal to the ESC among other things.
So: Nope.

duffann, 2019-10-21:
Thanks for the answer,

1) I Saw your post about Bluetooth cutting antenna,But it does not completely block the Bluetooth connection it just reduces the distance so is there anyway to completely block any Bluetooth connection to the board?

2) I have a confusion about flushing the Bluetooth boards and to main boards .my problem is I want to leave the Bluetooth with the most updated version so that the messages are encrypted so people cannot connect to Bluetooth but I want to flash the main board with your custom firmware created on your website so I tried to directly connect to main boards with ST links version two but I can only flush the recovery firmware  from here when I select the Bin file created on your website I get an error saying something about salt so is it possible to leave the Bluetooth bought it the latest version and just directly connect to main boards and Install the bin file created on your website ? I am using the m365rec.exe to flush the mb ,

gummyballz, 2019-11-18:
I made it work.   I took the control board from a lyft 365 and the comm board from a bird a viola.  Life