BotoX/xiaomi-m365-firmware-patcher #80: Asking Scooter Version, no response

crosisca, 2019-11-07:
I just got a brand new scooter (chinese black version), first time I connected it to MiHome it uploaded to 1.5.4.
When trying to use the DownG app it stays stuck at "Asking Scooter Version" when I click Check Version.
Does that mean I'm unable to to downgrade and use custom firmwares?

Daxh64, 2019-12-14:
Check on the internet. You can't flash past version 1.5.1 + so good luck finding a way to flash to the previous version (1.4.3 max) and flash it again with the customs firmware.

kdrapel, 2020-01-20:
I had similar issues, managed to downgrade and commented this on Youtube. The video I am refering to is: https://www.youtube.com/watch?v=nlhz8XJFMYg

_Starting point --> I had a BLE v091 and a DRV 155, BMS 116, they were updated via Xiaomi app when I bought the scooter. So I needed to downgrade to restore things and install the custom fw. I installed latest M365_DownG, then took the BLE bin for 072 (don't keep the 091). I flashed the board with this (connect, check ver, load binary, flash). Then I went to Xiaomi Mija M365 Custom Firmware toolkit page, picked the "Mine" settings though I unchecked Patch for "KERS" and the Battery voltage settings. I flashed this and my leds went back to a normal state. Check  the video 'flash custom firmware using M365_DownG (no need for ST-LINK)', it's a life-saver. I have a Pro clone board but it's bricked and am waiting for the ST Link to resurrect it._

roadrgo, 2020-03-22:
Anyone that knows how to repair Xiaomi m365? Kindly write to me if so.