BotoX/xiaomi-m365-firmware-patcher #82: Cannot Increase Max Speed with Custom Firmware

JosephSamMorris, 2020-01-11:
I upgraded my m365 to the m365 Pro dashboard recently, and used the Gonkad365 preset. Everything works as expected, except that the top speed is still limited to 27 kph in drive and sport mode. I've tested it in the air without any load, so my weight isn't a factor, and I tried with "enable built-in current-driving throttle algorithm."

I also double checked that the CFW was uploading correctly by dropping the top speed in drive mode to 15 kph, and when I tested it, the scooter limited the speed to 15. So the CFW's are working.

Only thing I can think of is that when I first got the m365 I hastily upgraded to the most current FW, but since then I've installed the Pro dash, downgraded the BLE, BMS, DRV, and uploaded CFW.

Thoughts/feedback?

gunnerysurgeon, 2020-01-18:
try changing the limit to 35 in normal mode, because IIRC sport mode does not work properly on the regular m365 as its not made for it

also keep in mind that there is a hardware speed limit of ~32km/h that you cannot bypass without hardware mods

stmcata, 2020-03-26:
> also keep in mind that there is a hardware speed limit of ~32km/h that you cannot bypass without hardware mods

Hi, 

What hardware speed limiter is there? Do you know what hardware mods are available?

Thanks.

danteodorescu, 2020-03-27:
Hello,

I have a Xiaomi Mi365 scooter and I can not increase maximum speed over 28km/h.
I have BLE Ver72, BMS  Ver 126 and custom DRV.
I have tried BMS Ver 115 and Ver 126 and many customs DRV (combination of Maximum speed in Normal mode 30, 32, 35, 36, Maximum speed in Eco Mode 18, 20, 30, 32, Motor power constant 51575, 40165, 45000, 38000, Remove hard speed limit of 35km/h On and Off, Russian throttle algorithm (from CFW) On and Off)
In all cases everything works exactly like I write in firmware except maximum speed which never exceeds 28 km/h. 
Every time I tested with the wheel lifted in the air.
What could I try?
Has anyone else had this problem?
How was it resolved?

stmcata, 2020-03-27:
Same here, that's why i'm asking for any modifications besides firmware.