BotoX/xiaomi-m365-firmware-patcher #90: Overvolt M365

LGM33, 2020-04-13:
What is the maximum voltage the controller will support? What is the max voltage override capable in firmware?

From my research, the controller main capacitor is 63v and the MOSFETs are 75v_max 80A STB75NF75.

I see people running 12s 50.4v_max without any controller modification.
10s3p(OEM battery)+2s3p(Additional battery) in series
A Russian guy 3D printed a 2s3p battery holder to run 12s.
https://www.thingiverse.com/thing:3714249
https://electro.club/forum/uvelichenie_napryajeniya_pitaniya_dvigatelya

I hear 13s 54.6v_max is possible, but can blow controller on regen brake. Maybe changing from a 63v capacitor to a 100v one will solve the issue?

Also, I saw someone using a BEC (Battery Eliminator Circuit) to power the BLE board with 5v when using higher voltage on the controller. Im not sure at what voltage that is necessary.

What is the max voltage the motors can handle? Using a Chinese 72v controller running on 20s 84v_max would double the top speed to 60km/h.

LGM33, 2020-04-26:
Edit: The MOSFETs are actually 100v_max 110A STP15810. Maybe changing out the 63v capacitor and the 5v supply circuit can enable running 84v_max battery?

LGM33, 2020-04-26:
I found a video from a Spanish guy that modified original m365 the controller to use two 36v batteries for 84v max.
https://www.youtube.com/watch?v=CfD5VDLyOi4

I also found this forum post detailing the steps:
https://scootertalk.org/forum/viewtopic.php?f=55&t=6694

1. Flash custom firmware with above 60km/h speed limit, remove KERS, and add above 84v override(unsure if needed).

2. Remove the ESC control board from housing.

3.On the ESC board, cut the Positive PCB trace going from the XT30 connector to the 63v Capacitor positive terminal.
(VERY IMPORTANT. This will isolate the STM32F Controller and 5v DC converters from the 84v battery supply. ESC and BLE boards will blow if not isolated.)

4. Remove the 63v capacitor on the control board.

5. Add a 100v capacitor to replace the 63v capacitor. Connect the negative lead to the negative capacitor pad and connect the positive lead to the positive of the XT30 input. (The positive capacitor pad should be disconnected from the positive XT30 input at this point)

6. Connect two 36v batteries in series using XT30 series connector harness and plug into XT30 input on ESC.

7. Connect the positive of the 1st 42v battery to the positive capacitor pad on ESC board. (This will supply 42v to power the MCU and 5v DC converters isolated from the main 84v battery pack.)

8? Replace the six MOSFETs with 100v 120A IRFB4110 MOSFETs (Maybe not needed. The default MOSFETs are already 100v 110A STP15810.)

Has anyone tried this method?

samko1234, 2021-06-06:
Why the video on youtube is private