BotoX/xiaomi-m365-firmware-patcher #91: M365 problens with the firmware update.

skizophrenic365, 2020-04-14:
Hi i am new to this so i will try to explain what i did. I just downgraded the BLE after that i used that webpage to make a firmware witch i did but it was that "spoofing version" so now when i am clicking on check version it's says dvr538 instead of dvr 138. Is it any chance to go back to v1.54 version of firmware? Thank you!

maurus69, 2020-06-22:
Hi, i have the same problem.. have you got any answer?

alphachris, 2020-07-21:
Hi there, I'm having the same problem. Someone figured out meanwhile how to "revert" back? Just leaving the patch option for "version spoofing" unchecked didn't do the trick unfortunately. Tried various combinations with 1.4.0 to 1.3.8 but no luck. Version displayed is still 5.x.x. Unfortunately changes to the FW are not accepted either, e.g. trying to unpatch russian throttle algo remains on the BLE.
Do you have any idea what else we could try @BotoX ?

alphachris, 2020-07-21:
Okay, nevermind. I found the solution on my own.
Anyway, I'd like to share my finding with everybody who's stuck in the same situation that I was (M365 classic - don't know if this solution also works for M365 Pro).

1. uninstall M365 DownG (if installed).
2. download and install the patched M365 DownG App (V7 intended for developers) from here: https://cloud.botox.bz/s/PZkfxWjNpgXibT5/download?path=%2F&files=com.m365downgrade-v7_PATCHED.apk - this version allows to flash unencrypted .bin CFW's no matter what - so use at your own risk and pay some extra attention!
3. go to https://m365.scooterhacking.org and create 1.3.8 with Rollerplausch preset from the top, but make sure to uncheck "Version Spoofing"! (Otherwise you'll most probably brick or at the very least find your scooter in the same state as it was before). Before clicking "Patch" at the bottom of the page, set output file to "Bin".
4. Open the patched M365 DownG app, connect to your scooter and flash the previously downloaded .bin file.
You should now see DRV 1.3.8 installed again (at least for me it worked). Hope this helps!