BotoX/xiaomi-m365-firmware-patcher #100: Xiaomi essential scooter support

szcsongor, 2020-11-06:
I got the essential scooter which I believe is essentially (no pun intended) an M365 Lite.
It's firmware is 2.1.8 and it has three speed modes (Eco, Drive, Sport), not two (Eco, Normal).
How to proceed then to build a custom firmware at https://m365.botox.bz/?
Or is it not supported yet?

Adorfer, 2020-11-06:
if "no pun intended": please make the topic more specific. 
For my understand the firmwarepatcher currently is actually essential.

szcsongor, 2020-11-06:
Don't really understand you. (Or you don't get the meaning of "no pun intended". If not, just ignore it...)
I think I was specific, but I can elaborate.

What I have: Xiaomi Essential (which is listed as M365 Lite in custom control apps like m365 tools)
What I wanted: a custom firmware to boost speed
What I see at https://m365.botox.bz/: selectable firmwares 1.3.8 and 1.4.x and speed modes eco and normal
What I have: firmware 2.1.8 and speed modes eco, drive and sport

This discrepancy made me wonder if I should continue (or is it even possible) to build a firmware or not.
Sorry, I'm new to scooters and this thing, might be that I miss something here.

Adorfer, 2020-11-06:
sorry for asking that sclarificatin. not knowing about "Xiaomi essential" beeing a model, it was not unclear for the issue that you wanted support for a new model.

szcsongor, 2020-11-06:
No problem, I appreciate your reply and willingness to help.
Yes, it's a new model, but I think they just repackaged or upgraded the M365 lite.
So I guess the support would not be a problem, only its base firmware should be added and the slightly different features (speed modes) added.

Grizaq, 2021-07-05:
Anyone has same issue /any updates on the case?
I can confirm that the versions are (still) 1.3.8 - 1.4.3 in the patcher, while mi m365 essential has the 2.1.8 version (I believe that's still the same version released with the scooter's launch).
Also doesn't seem like the patcher is suitable, as the scooter has a different initial speed value (the patcher has 22 and 28 km/h default values, while the scooter limits are 15 and 20 (not mentioning this model has 3 speed options))
In any case, does anyone know or have tried doing it yet?
Highly appreciate any responses.

Grizaq, 2021-07-10:
So, in case someone still needs this thread:
Pro 2 firmware seems to be compatible with s1 and essential, but can't affect the walk assist mode (worked this way for my essential at least), but in my case a bug appeared, cruise conto not working, trying to troubleshoot it now