BotoX/xiaomi-m365-firmware-patcher #102: Is it possible to revert back to the original?

tifDev, 2020-11-12:
Hello,

Congratulations on this app.

I want to try it but before, I want to know if there is way to backup the current firmware before patching so that I can revert back if I want to.

Is this possible?

Also, I have a m365 classic/normal (In the model it says just M365 so I think it's the classic) , does it works on this?

Thank you

CommanderRedYT, 2021-03-14:
You just have to update the firmware in the Xiaomi Home App.
It works fine on my standard M365 and my friends M365 Pro

tifDev, 2021-03-29:
I don't think I will do that because I heard some things about updating the firmware with Xiaomi (Like speed lock in EU).

But thanks :)

closed 2021-03-29 by tifDev

CommanderRedYT, 2021-04-03:
What I just noticed is that you can also click on default on the botox website and thus return to the original software