BotoX/xiaomi-m365-firmware-patcher #103: Is it possible to revert back to original?

tifDev, 2020-11-12:
Hello,

Congratulations on this app.

I want to try it but before, I want to know if there is way to backup the current firmware before patching so that I can revert back if I want to.

Is this possible?

Also, I have a m365 classic/normal (In the model it says just M365 so I think it's the classic) , does it works on this?

Thank you

CommanderRedYT, 2021-03-14:
You just have to update the firmware in the Xiaomi Home App

dave90dave, 2021-06-12:
> You just have to update the firmware in the Xiaomi Home App

I have flashed a image of the M365 on my M365 PRO 1 scooter.
Since the model was different, how can I revert this firmware?
Because the M365 has less options then the M365 PRO.