BotoX/xiaomi-m365-firmware-patcher #107: Compatibility with hover-1

notsolowki, 2021-01-27:
I see people saying the hover-1 alpha is very similar to the m365. Could this firmware for the hover-1 alpha?  I have dissembled a hover 1 alpha and have found rx,tx and checked various wires with my oscilloscope.

 I was able to manipulate the speedometer with a frequency generator but it wont go higher than 18mph @360mhz. I think there might be 2 processors in this scooter.

 If any interest in this scooter I could have plenty of pictures of the hardware. 

When I probed the communication wire from the processor on the handle bars for speedometer and throttle I found what looked like some data transmission not just pwm.

bluegizmo83, 2021-04-03:
> I see people saying the hover-1 alpha is very similar to the m365. Could this firmware work for the hover-1 alpha? I have dissembled a hover 1 alpha and have found rx,tx and checked various wires with my oscilloscope.
> 
> I was able to manipulate the speedometer with a frequency generator but it wont go higher than 18mph @360mhz. I think there might be 2 processors in this scooter.
> 
> If any interest in this scooter I could have plenty of pictures of the hardware.
> 
> When I probed the communication wire to the esc from the processor on the handle bars for speedometer and throttle I found what looked like some data transmission not just pwm.

Did you try flashing this firmware to see if it works?

68pishta, 2023-07-29:
I got no power function out of the power switch, I confirmed the harness is good and the battery is at 38V. I have 38V on the EN pin of the small dash PCB but I do NOT have 12V on the pin next to it. Where does this 12V power come from?